microsoft/vscode-arduino #1104: Can't Select Serial Port

rhit-jacksocr, 2020-09-19:
Issue Type: <b>Bug</b>

I cannot select my serial port. I have tried every fix I can find on google and nothing works

Extension version: 0.2.29
VS Code version: Code 1.43.2 (0ba0ca52957102ca3527cf479571617f0de6ed50, 2020-03-24T07:38:38.248Z)
OS version: Windows_NT x64 6.1.7601

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz (8 x 2494)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>viz_display_compositor: enabled_on<br>viz_hit_test_surface_layer: disabled_off_ok<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.88GB (1.04GB free)|
|Process Argv||
|Screen Reader|no|
|VM|0%|
</details>

PeterWone, 2020-09-20:
Duplicate of #1098

sukesh-ak, 2020-09-20:
> Duplicate of #1101, #1099, #1098, #1097

Which one should be subscribe to for updates?

simonl65, 2020-09-20:
As a workaround:

1. open the folder where your .ino file resides
2. open the .vscode folder
3. edit arduino.json
4. change the port 
e.g.
```
{
  "port": "COM3"
}
```

simonl65, 2020-09-21:
Related issues: #922 #1105 #1104 #1101 #1099 #1098

sukesh-ak, 2020-09-23:
> As a workaround:
> 
> 1. open the folder where your .ino file resides
> 2. open the .vscode folder
> 3. edit arduino.json
> 4. change the port
>    e.g.
> 
> ```
> {
>   "port": "COM3"
> }
> ```

That's working only for flashing but not for serial monitor

Abneco, 2020-09-25:
This trick didn't work for me, I uninstalled the Visual Studio code, and reinstalled an old version, it was no longer a problem and disabled automatic updates ...

https://code.visualstudio.com/updates/v1_41

"update.mode": "none"

fabnicolas, 2020-10-05:
@Abneco This workaround works for me but does it exists a portable version of Visual Studio Code?

Right now because of this extension I installed "Visual Studio Code" old version specifically to use vscode-arduino and "Visual Studio Code Insider" for other projects, but I really wanted to simply have a portable vscode designed only for Arduino projects, but I can't find it so far.

Would be nice if this issue is fixed tho to avoid this workaround.

PeterWone, 2020-10-05:
@fabnicolas I looked into doing this.

The issue is not in the Arduino extension but in one of its _indirect_ dependencies. A rewrite to use the N-API indirect invocation of native libs is far from trivial due to the complexity of post-installation platform specific build (on which the current implementation relies).

I am currently building an alternative extension that does not suffer from this problem because it works in a completely different way, shelling command line utilities for native integration. This has been a great deal of work and I have not yet decided to make it available for free.

Abneco, 2020-10-05:
Personally, I also have to develop on a stable system, I find that this extension is promising but not mature enough, so I chose to work with platformIO https://platformio.org/ compatible with the latest version and much more stable , easy to configure ... PlaformIO has another small drawback, ... acceptable ... this is mainly due to the fact that you must respect c ++

sukesh-ak, 2020-10-06:
> @Abneco This workaround works for me but does it exists a portable version of Visual Studio Code?
> 
> Right now because of this extension I installed "Visual Studio Code" old version specifically to use vscode-arduino and "Visual Studio Code Insider" for other projects, but I really wanted to simply have a portable vscode designed only for Arduino projects, but I can't find it so far. Telling my team to use a specific version of VSCode on their machine by uninstalling their ones is not acceptable.
> 
> Would be nice if this issue is fixed tho to avoid this workaround.

There is a new project called Arduino Pro which is trying to build VS Code kind of IDE. But I would not jump personally for now.

sukesh-ak, 2020-10-06:
Also Arduino has a new project for building  a new command line. I guess that will become the main way to compile, flash etc in the future.

CorentinPacaud, 2020-10-15:
Will this issue be fixed some day ? or just end of support ?

kseyhan, 2020-10-20:
try rebuilding the plugin following the steps from https://github.com/microsoft/vscode-arduino/issues/1113#issuecomment-713100975.

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

robotdad, 2020-10-30:
Resolved in release 0.3.3

closed 2020-10-30 by robotdad

WenukaB, 2020-11-01:
> As a workaround:
> 
> 1. open the folder where your .ino file resides
> 2. open the .vscode folder
> 3. edit arduino.json
> 4. change the port
>    e.g.
> 
> ```
> {
>   "port": "COM3"
> }
> ```

This prompts the error "The arduino.json file format is not correct."

WenukaB, 2020-11-01:
This can probably be due to an issue with Arduino Installation. I had the same issue. I uninstalled the Arduino existing installation and installed the latest version (1.8.13) it fixed the issue.

simonl65, 2020-11-01:
This is now fixed with vscode-arduino 0.3.3

ryanlightbrighton, 2021-02-17:
Just installed this (v 0.3.4).  I can't select serial port.  I added com3 manually to the arduino.json but then I'm stuck on the other bug (can't open serial port)

Lomolisso, 2021-02-17:
i have the same problem @ryanlightbrighton

adiazulay, 2021-02-17:
This is resolved in v0.3.5 see #1198 for installation instructions.