microsoft/vscode-arduino #1110: What am I Missing?

julesgilson, 2020-10-03:
As I was installing VSC on a new laptop I thought I would give this extension a go.... After playing around with include paths for a few hours I still cannot get it to either:

- Find the Arduino core includes (Serial, types, etc etc)
- Find the defines/methods from includes in the workspace folder
- Find the methods in separate sketch files in the workspace folder (they should be included by default)
- Get any kind of relevant code completion - only generic C++ suggestions

At this stage it seems better/less frustrating to stick with Arduino IDE - and that isn't something I thought could be possible.

What am I doing wrong? Windows 10 2004 - VSC 1.49.3 - Arduino IDE 1.8.13 and why aren't a reasonable set of platform defaults included for the cpp properties? There isn't even a mention of having to config the cpp properties in the extension page on the Marketplace!

adiazulay, 2020-11-20:
We should have a release candidate tomorrow that fixes some of the intellisense issues

Arcturuss, 2020-11-30:
@julesgilson it seems, this extension need some config files on top of current open folder to work properly. These can be generated with Ctrl+Shift+P -> Arduino:Initialize

julesgilson, 2020-11-30:
> @julesgilson it seems, this extension need some config files on top of current open folder to work properly. These can be generated with Ctrl+Shift+P -> Arduino:Initialize

That performs the same function as selecting a board type.

Most of the issues I have experienced with this plugin have been due to it not correctly building the initial cpp config (i.e related to intellisense - but if not for that might as well use Arduino IDE!) For a start I would expect that architecture to be GCC not MSVC as the Atmel compiler which will end up being used is a GCC compiler. I expect it defaults to GCC on a Linux machine, on Windows it is MSVC.

Additionally, although it inserts some includes for the ESP8266 library when you select such a board - it doesn't include the variant specific header, so things like pins defs are missing.

The biggest single problem with the extension is the lack of documentation. After some trial and error I have found out that you can use header files (to satisfy intellisense) so long as you name them very carefully. If all this was documented it would not be an issue.

And, yes, adding USBCON to the default cpp config does fix the link to Serial. But that was just one of the issues I raised.

I hope that this extension keeps getting better. I think it could be the best solution for general Arduino development.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay