microsoft/vscode-arduino #1111: Arduino Debugging Environment not displaying in list

DerekFoster1976, 2020-10-03:
I use an MXChip (with VSC 1.48.2) which has fully supported debugging with the extension. However, when I just have the .ino file open in the workspace (no other files open) and try to debug by Run>Start Debugging it does not list the Arduino environment (see screenshot1), When I open an additional file such as the settings.json file the full list of environment options is shown when I do Run>Start Debugging (screenshot2).

[image]

[image]

jimbobbennett, 2020-10-03:
I also see the same when trying to create a launch.json file from VSCode. If I have the sketch open, I only get the C++ options, I have to have the settings.json open to be able to get a launch.json created with the Arduino debugging option.

fredsa, 2023-08-08:
This issue may be related to #1632, which reports breakage as of `v0.6.0`.