microsoft/vscode-arduino #1112: Arduino IDE cannot be found

FarmerHans, 2020-10-06:
Hello i wanted to start programming an Arduino in Visual studio Code. After I installed the Extension and the IDE and several tries I didn't find a solution to this problem. I also tried to set the path in the settings.json.  and i tried reinstalling the ide and visual studio code still no solution. 
Thx Tom

HEMISixty, 2020-10-10:
Try googling for the command line instructions for the IDE.  I think this helped resolve a similar issue someone else had.  I think this is the link...

https://learn.sparkfun.com/tutorials/efficient-arduino-programming-with-arduino-cli-and-visual-studio-code/all

T0mHutter, 2020-10-12:
Ok but this it not the awnser for my question i wanted to use the arduino extension especially and the article is about the c/ c++ extension 
thanks for the effort though.

HEMISixty, 2020-10-13:
My mistake, I thought you were having problems with getting the CLI piece to work.

T0mHutter, 2020-10-13:
No problem i just fixed it
but thanks

closed 2020-10-13 by T0mHutter