microsoft/vscode-arduino #1113: Can't Select Serial Port

CodeNameHawk, 2020-10-08:
Issue Type: <b>Bug</b>

Can't Select Serial Port

Extension version: 0.3.2
VS Code version: Code 1.49.3 (2af051012b66169dde0c4dfae3f5ef48f787ff69, 2020-10-02T17:56:11.027Z)
OS version: Linux x64 5.4.0-48-generic

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Pentium(R) Dual-Core CPU       T4500  @ 2.30GHz (2 x 2100)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: disabled_off<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: disabled_software<br>skia_renderer: enabled_on<br>video_decode: unavailable_off<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|1, 1, 1|
|Memory (System)|3.81GB (1.22GB free)|
|Process Argv|--no-sandbox --unity-launch --crash-reporter-id 02ee2987-17b5-4cfb-9854-12b20aba17e2|
|Screen Reader|no|
|VM|0%|
|DESKTOP_SESSION|xubuntu|
|XDG_CURRENT_DESKTOP|XFCE|
|XDG_SESSION_DESKTOP|xubuntu|
|XDG_SESSION_TYPE|x11|
</details>

zhengzichun, 2020-10-08:
I have similar issue.

Version: 1.49.3 (system setup)
Time: 2020-10-02T17:57:01.595Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 6.1.7601

CPUs | Intel(R) Core(TM) i5-2540M CPU @ 2.60GHz (4 x 2594)

GPU Status | 2d_canvas: enabled flash_3d: enabled flash_stage3d: enabled flash_stage3d_baseline: enabled gpu_compositing: enabled multiple_raster_threads: enabled_on oop_rasterization: unavailable_off opengl: enabled_on protected_video_decode: unavailable_off rasterization: unavailable_off skia_renderer: disabled_off_ok video_decode: enabled vulkan: disabled_off webgl: enabled webgl2: enabled

Extension | Author (truncated) | Version
vscode-language-pack-zh-hans | MS- | 1.49.2
python | ms- | 2020.9.114305
cpptools | ms- | 1.0.1
vscode-arduino | vsc | 0.3.2

cdelcristo, 2020-10-10:
is anyone even looking to this issues???????

ayushsharma82, 2020-10-10:
Same Issue. Started happening after an update came to our Visual Studio Code.

kevtan, 2020-10-10:
Same issue, but on a mac!

RiverShi, 2020-10-11:
Every update of vscode bring more bug to arduino? Maybe 1.42 is good for vscode-arduino.

rrakso, 2020-10-14:
Still not working :<

Seefer, 2020-10-14:
I'm trying to follow the IoT DevKit tutorial for the mxchip device and running into this issue of simply trying to select the COM port too. This is not a beginner-friendly state for an extension to be in. It's been suggested I try PlatformIO until the problems with this extension are resolved.

homechicken, 2020-10-14:
Same issue here, on both MacOS and Windows 10: I can't select a port (although I can specify it in the arduino.json file). Also, it errors when I try to open the serial monitor.

alex-maha, 2020-10-16:
Broken here too.

Version: 1.50.1 (user setup)
Date: 2020-10-13T15:06:15.712Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

Also, clicking on the Open Seriual Monitor gives the following error:

Failed to open serial port COM1 due to error: + Error: The module '\?\c:\Users\xxxxxx.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 75. This version of Node.js requires NODE_MODULE_VERSION 80. Please try re-compiling or re-installing the module (for instance, using npm rebuild or npm install).

Emergency1999, 2020-10-17:
Same issue here. I can still overwrite the COM via the file, but the serial port still won't connect.

StealthAdder, 2020-10-17:
Same Here Cant Connect to Serial Port from the interface.

kseyhan, 2020-10-20:
Hi,
for everyone suffering this i figured out how to manualy build the plugin to make everything work with the latest version of vscode.

To build the plugin do the following:

     git clone https://github.com/microsoft/vscode-arduino
     cd vscode-arduino
     npm i
     npm i -D electron-rebuild
     ./node_modules/.bin/electron-rebuild -v9
     npm i -g gulp
     ./node_modules/.bin/gulp build
     rename the out folder in the vscode plugins directory to have a backup (~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out)
     copy over the "out" folder from the freshly build folder over to the plugins folder mentioned above

enjoy!

HEMISixty, 2020-10-20:
> Hi,
> for everyone suffering this i figured out how to manualy build the plugin to make everything work with the latest version of vscode.

Working on this now...thanks for posting!

cdelcristo, 2020-10-20:
> Hi,
> for everyone suffering this i figured out how to manualy build the plugin to make everything work with the latest version of vscode.
>
> To build the plugin do the following:
>
> ```
>  git clone [...]

It work Thanks

boskoe, 2020-10-21:
@kseyhan You sir are a steely-eyed missile man!

kseyhan, 2020-10-21:
> > Hi,
> > for everyone suffering this i figured out how to manualy build the plugin to make everything work with the latest version of vscode.
> 
> Working on this now...thanks for posting!
> 
> Update: That didn't work...I use this as a hobby and I'm not a programmer. I followed the steps using a linux machine. But, VSC is installed on my Windows 10 machine. Copying the output folder from one to the other apparently doesn't work...but, it was worth a shot...

@HEMISixty Its no wonder it doesnt work when you build elf binaries for your windows machine. You have to install nodejs on your windows machine and build it there if you want to to use the ide on windows.

HEMISixty, 2020-10-21:
Yeah, like I said, it was worth a shot.  If someone were to post a set of steps, like you did, for Windows, that would be awesome.

Like I said, this is an LED hobby for me and I'm not a developer.  But, what the heck, I suppose I could give it all a shot to figure out and post the steps if I'm successful.

Then again, I'm tempted to build a Linux machine and go that route instead...

Anyway, your steps are beneficial to some folks and it's cool of you to post a solution.

Cheers!

kseyhan, 2020-10-21:
@HEMISixty the commands should all be the same for windows. you maybe have to adjust the slashes to fit the windows convention thats should be it.

HEMISixty, 2020-10-21:
Cool, I'll give it a shot then.  Thanks...

RBM-PITT, 2020-10-21:
When running in Windows - make sure you run as administrator.

Also, this could be a problem depending on your build of Windows:
Error: Node Sass does not yet support your current environment: Windows 64-bit with Unsupported runtime (88)

kseyhan, 2020-10-21:
by a quick search the solution to this seems to rebuild the sass module aswell.
`npm rebuild node-sass`

RBM-PITT, 2020-10-21:
@kseyhan thanks - I'll try that!

boskoe, 2020-10-21:
Go team go!

RBM-PITT, 2020-10-21:
@boskoe Love the support!
@kseyhan I rebuilt the dependencies successfully for NPM, still get this error
[13:36:04] Using gulpfile ~\vscode-arduino\gulpfile.js
[13:36:04] Starting 'build'...
[13:36:04] Starting 'clean'...
[13:36:04] Finished 'clean' after 136 ms
[13:36:04] Starting 'ts-compile'...
[13:36:11] Finished 'ts-compile' after 6.74 s
[13:36:11] Starting 'html-webpack'...
[13:36:33] Error (webpack): ./app/styles/board.scss
Module build failed (from ./node_modules/mini-css-extract-plugin/dist/loader.js):
ModuleBuildError: Module build failed (from ./node_modules/sass-loader/lib/loader.js):
Error: Node Sass does not yet support your current environment: Windows 64-bit with Unsupported runtime (88)

boskoe, 2020-10-21:
@RBM-PITT Did you do a rebuild of Sass as was done to the electron node?

RBM-PITT, 2020-10-21:
@boskoe I did the rebuild as @kseyhan called out : npm rebuild node-sass

If i do it with the "npm i -D node-sass" it fails on a path error.

npm ERR! code 1
npm ERR! path C:\Users\robma\vscode-arduino\node_modules\node-sass
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c "node scripts/build.js"
npm ERR! Building: C:\Program Files\nodejs\node.exe C:\Users\robma\vscode-arduino\node_modules\node-sass\node_modules\node-gyp\bin\node-gyp.js rebuild --verbose --libsass_ext= --libsass_cflags= --libsass_ldflags= --libsass_library=

kseyhan, 2020-10-21:
sry, cant help you there. i just tried to reproduce it on windows right now but had to abort since i didnt had enough free space on my harddrive to install visual studio and the other dependencies that are needed to compile binaries on windows. someone else has to come up with the solution i guess.

boskoe, 2020-10-21:
You two are far past my programming skills. I did a bit of Googling though and came up with this bit of distressing information.

[node-sass doesn't support being run in in Electron (VS Code) versions of node, so we can't help troubleshoot](https://github.com/sass/node-sass/issues/2962#issuecomment-685227551)

boskoe, 2020-10-21:
Reading the related issues to this on the vscode github indicates that no fix will come from within vscode.

HEMISixty, 2020-10-21:
This sounds like I shouldn't bother trying to update my Windows machine and either switch to a linux machine or just configure it using the json file...

boskoe, 2020-10-21:
Linux seems to work fine with the fix presented  by @kseyhan. Even on a RaspberryPi 4. So I'm going to do just that as there are some very nice improvements in the later versions of vscode and C++.

HEMISixty, 2020-10-21:
Yeah, I think I'll be doing the same thing...

kseyhan, 2020-10-22:
@RBM-PITT ok then ive got a even hackier solution for you
just rebuild the usb-detection and the bindings module and overwrite the binaries in the extensions folder.
(adjust the paths to your envinronment!)

git clone https://github.com/microsoft/vscode-arduino
cd vscode-arduino
npm i
npm i -D electron-rebuild
./node_modules/.bin/electron-rebuild -v9 -o usb-detection,bindings
cp ./node_modules/usb-detection/build/Release/detection.node ~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/usb-detection/build/Release/detection.node
cp ./node_modules/@serialport/bindings/build/Release/bindings.node ~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/@serialport/bindings/build/Release/bindings.node

trullock, 2020-10-22:
@kseyhan After completing this I get

```
Failed to open serial port COM6 due to error: + Error: \\?\c:\Users\Andrew Bullock\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application. \\?\c:\Users\Andrew Bullock\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node
```

kseyhan, 2020-10-22:
trullock have you build that binary on a windows machine or did you copy it from a linux machine over?

trullock, 2020-10-22:
@kseyhan built on the same windows machine as VSC, using node 12.19.0 x32

kseyhan, 2020-10-22:
grrrrr... i had to uninstall my game from my windows machine to test that now.. ( you guys owe me a beer!)
@trullock it works for me

maybe something is wrong with your vsc.

run
npm install --global windows-build-tools

from an elevated cmd prompt before running the commands.

PeterWone, 2020-10-22:
Don't bother. I've already finished feature matching and shortly I'll be putting out an arduino extension that

* doesn't break every time VS Code changes Node version
* doesn't have a dependency on the Arduino IDE
* downloads and installs all the dependencies that it _does_ have (Arduino CLI and the appropriate "core" (toolchain and libs and headers) for your platform and board (OS and Arduino type).

I'm not going to feature match debugging because most users don't have the specialised hardware for it.
I _am_ going to manage the search paths so that intellisense works.

Search path management is the only outstanding item. 

I do need volunteers for testing.

kseyhan, 2020-10-22:
@PeterWone why arent you extending this extension instead of creating a complete new one?

kseyhan, 2020-10-22:
@trullock 
another option might be to simply specify your arch when running the electron-rebuild.

`./node_modules/.bin/electron-rebuild -v9 -o usb-detection,bindings -a yourarchnamehere`

PeterWone, 2020-10-22:
@kseyhan because I did everything completely differently and reached feature parity in 400LOC

kseyhan, 2020-10-22:
@PeterWone uhm, yes well... i'm curious how it turns out then.

PeterWone, 2020-10-22:
@kseyhan The Arduino extension from the Microsoft China team (this repo) uses undocumented interfaces to call into the Arduino IDE, that's how it compiles and deploys. My stuff uses documented supported CLIs. I think it has more future. USB serial device enumeration is slower but it's not even a first release yet. I have had input from the Arduino CLI dev team which certainly sped development up. It started out as a proof of concept but ten days later I realised that apart from debugging I had feature parity.

trullock, 2020-10-22:
@PeterWone +1 for your approach, let us know where to post the beer

PeterWone, 2020-10-22:
Thank you, but what I needis testers and feedback. My other extension has a perfect record of five star reviews but this code has very large surface making QC a hell of a job. I can only test Windows. I need volunteers representing other platforms, _especially_ once I build the path management bits.

RBM-PITT, 2020-10-22:
@PeterWone I can test Windows 10 & Ubuntu

PeterWone, 2020-10-22:
@RBM-PITT smashing, I'll get set up (it's not on github at the moment) and get back to you with the details.

trullock, 2020-10-22:
I can test on two diff Win10

boskoe, 2020-10-22:
@PeterWone I can test Windows 10 x64 and Debian 10

PeterWone, 2020-10-22:
OK volunteers, let's move the discussion.
https://github.com/PeterWone/vscode-arduino-cli

trullock, 2020-10-22:
Oh lawd if you fix the C++ intellisense I will literally ship you beer from the UK to AU

HEMISixty, 2020-10-22:
Volunteered!

PeterWone, 2020-10-22:
People, please lets move this to the new repo. Partly for our own convenience and partly out of respect to MS China who got us this far.

maddogjt, 2020-10-27:
I've published a rebuild of v0.3.2 with the serial packages rebuild on the proper interface for anyone who needs.   https://github.com/maddogjt/vscode-arduino/releases/download/v0.3.2%2Bserialfix/vscode-arduino-0.3.2+serialfix.vsix

trullock, 2020-10-27:
@maddogjt nice work, this works for me

srevid, 2020-10-27:
@maddogjt work for me too, thanks !

CodeNameHawk, 2020-10-27:
"I've published a rebuild of v0.3.2 with"
not work.

closed 2020-10-27 by CodeNameHawk

trullock, 2020-10-27:
@CodeNameHawk Why have you closed this? 0.3.2 is still broken.

CodeNameHawk, 2020-10-27:
> 
> 
> @CodeNameHawk Why have you closed this? 0.3.2 is still broken.

Try manually uninstall  0.3.2, and install  0.3.2+serialfix from link above.

New you see plugin name 0.3.2+serialfix.

trullock, 2020-10-27:
@CodeNameHawk sorry, just to be clear on my part, there is a bug with the Microsoft `vscode-arduino` v0.3.2 extension for Microsoft Visual Studio Code, meaning it doesn't work. 

And the official response is to hope that the 808263 people who have it installed will happen across a comment deep in a thread containing mostly a discussion about making a new extension as a result the communities opinion on how this one is maintained?

You're not going to actually fix this and publish a 0.3.3 to the VSC "extension store" for everyone to benefit from?

Or am I mistaken

[9 more replies]