microsoft/vscode-arduino #1116: Vs Code Arduino Extension <Select Serial Port> doesen't work

T0mHutter, 2020-10-18:
My select a ... doesn't work i just reinstalled the Extension as well as the recommended versions of vsc in the other issues. In the Arduino IDE the select a Serial Port works perfectly but in vsc it doesen't

mansdahlstrom1, 2020-10-18:
Got the same issue here, when clicking <Select Serial Port> nothing happens, i can go around this though by adding port manually in the `arduino.json` file like this

`"port": "/dev/cu.usbserial-xxxxxxxx"`

if I (after chaning port manually) this press the button next to it (where i normally change baud rate) i get this: 
```
Failed to open Serial port due to error: + Error: dlopen(/Users/mansdahlstrom/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/usb-detection/build/Release/detection.node, 1): no suitable image found. Did find: /Users/mansdahlstrom/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/usb-detection/build/Release/detection.node: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00 /Users/mansdahlstrom/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/usb-detection/build/Release/detection.node: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00
```

Running 
- Mac OS Catalina 10.15.7
- Arduino 1.8.13
- VsCode Arduino 0.3.2

T0mHutter, 2020-10-18:
yea got the same error
but im on windows so 
windows 10 newest version

wi1k1n, 2020-10-18:
Confirm a problem.

My setup:

Version: 1.50.1 (user setup)
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T15:06:15.712Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.18363

When I click on <span><</span>Select Serial Port> (or execute >Arduino: Select Serial Port) I get the following message in console:
```
console.ts:137 [Extension Host] [Warn] no library available after trying files (3) ["c:/Users/vfpkj/.vscode/extensions/vsciot-vscode.vs…native/lib/native/detector_darwin_7.1.11_x64.node", "c:/Users/vfpkj/.vscode/extensions/vsciot-vscode.vs…e/lib/native/detector_Ubuntu14.04_7.1.11_x64.node", "c:/Users/vfpkj/.vscode/extensions/vsciot-vscode.vs…-native/lib/native/detector_win32_7.1.11_x64.node"]0: "c:/Users/vfpkj/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/node-usb-native/lib/native/detector_darwin_7.1.11_x64.node"1: "c:/Users/vfpkj/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/node-usb-native/lib/native/detector_Ubuntu14.04_7.1.11_x64.node"2: "c:/Users/vfpkj/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.2/out/node_modules/node-usb-native/lib/native/detector_win32_7.1.11_x64.node"length: 3__proto__: Array(0)
```

T0mHutter, 2020-10-19:
i Have to add that i only got the issue when i manually set the port in the json file. When i don't set it i cant even select one.

kseyhan, 2020-10-20:
maybe try a manual build aslong there is no new version beeing published.
check https://github.com/microsoft/vscode-arduino/issues/1113#issuecomment-713100975

T0mHutter, 2020-10-21:
Where are you building the plug in

maddogjt, 2020-10-27:
I've published a new build here: https://github.com/maddogjt/vscode-arduino/releases/download/v0.3.2%2Bserialfix/vscode-arduino-0.3.2+serialfix.vsix

joem-msft, 2020-10-28:
Adding another view of the problem. When I try to connect I get the following error:
```
Failed to open serial port COM4 due to error: + Error: The module '\\?\c:\Users\joemay\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 75. This version of Node.js requires NODE_MODULE_VERSION 80. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).
```

T0mHutter, 2020-10-29:
I wanted to download maddogjt's build but it does not work. 
Does some one else downloaded

randomeizer, 2020-10-29:
It worked perfectly for me.

T0mHutter, 2020-10-29:
ok did you just downloaded it and it worked or is there any further steps that i did not make so it wont work

T0mHutter, 2020-10-29:
i got the error install failed

randomeizer, 2020-10-29:
Yep, I just downloaded and installed via the "Install from VSIX" command.

T0mHutter, 2020-10-29:
ok 
but i did not work for me

wi1k1n, 2020-10-29:
> Yep, I just downloaded and installed via the "Install from VSIX" command.

Works for me, thanks for the explanation!

T0mHutter, 2020-10-29:
do you have the latest version installed 
or an older one

wi1k1n, 2020-10-29:
> do you have the latest version installed
> or an older one

I have version 1.50.1  (the one that I installed from .exe file from oficial website).

The only thing is that I uninstalled the previous version of VSCose-Arduino plugin first, then reloaded VSCode and then installed the "+serialFix" version using [Extensions] -> [...] -> Install from VSIX...  menu

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

T0mHutter, 2020-10-29:
now it worked thx.

robotdad, 2020-10-30:
Resolved in the 0.3.3 release

closed 2020-10-30 by robotdad