microsoft/vscode-arduino #1117: not open serial port

gaston-cb, 2020-10-18:
Issue Type: <b>Bug</b>

Failed to open serial port COM10 due to error: + Error: error: 193\\?\c:\Users\gaston\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node 
is problem VScode, with open serial port different app (arduino IDE, and hercules) no problem with serial port 

Extension version: 0.3.2
VS Code version: Code 1.50.1 (d2e414d9e4239a252d1ab117bd7067f125afd80a, 2020-10-13T15:06:15.712Z)
OS version: Windows_NT x64 10.0.18363

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|AMD A10-9700 RADEON R7, 10 COMPUTE CORES 4C+6G  (4 x 3500)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|5.95GB (2.64GB free)|
|Process Argv|--crash-reporter-id 04018eee-e082-46c9-a6e6-352799f27541|
|Screen Reader|no|
|VM|0%|
</details>

kseyhan, 2020-10-20:
Check my https://github.com/microsoft/vscode-arduino/issues/1113#issuecomment-713100975 on how to rebuild the plugin.

maddogjt, 2020-10-27:
I've published a rebuild of v0.3.2 with the fixes - https://github.com/maddogjt/vscode-arduino/releases/download/v0.3.2%2Bserialfix/vscode-arduino-0.3.2+serialfix.vsix

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

HEMISixty, 2020-10-30:
@robotdad Does this adapt to future Electron updates and the like?

robotdad, 2020-10-30:
No, there are native packages for USB here that have to be updated when Electron updates. I'll see if we can come up with any ideas to make this more adaptable or not, if so I'll put something on the backlog.

robotdad, 2020-10-30:
Resolved in the 0.3.3 release

closed 2020-10-30 by robotdad

H4NNE5, 2021-02-15:
After the arduino extension (0.3.4) not working out of nowhere, I deleted VS Code completely and reinstalled ist.
Now I have this problem with the serial port.
Any advice?