microsoft/vscode-arduino #1119: cannot select serial port

dude-dou, 2020-10-27:
Issue Type: <b>Bug</b>

I cannot select the serial port, when I click on the "Open Serial Port" button, this is what I got:

Failed to open serial port COM4 due to error: + Error: \\?\c:\Users\<PCNAME>\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application.
\\?\c:\Users\<PCNAME>\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.2\out\node_modules\usb-detection\build\Release\detection.node

Extension version: 0.3.2
VS Code version: Code 1.50.1 (d2e414d9e4239a252d1ab117bd7067f125afd80a, 2020-10-13T15:06:15.712Z)
OS version: Windows_NT x64 10.0.19042

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz (8 x 2112)|
|GPU Status|2d_canvas: enabled<br>flash_3d: enabled<br>flash_stage3d: enabled<br>flash_stage3d_baseline: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: disabled_off<br>opengl: enabled_on<br>protected_video_decode: enabled<br>rasterization: enabled<br>skia_renderer: disabled_off_ok<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.85GB (0.61GB free)|
|Process Argv|--crash-reporter-id 06f577b2-fe72-4ce2-9d1c-47d9fe0091fe|
|Screen Reader|no|
|VM|0%|
</details>

maddogjt, 2020-10-28:
There are dozens of bugs already open for this issue, and it appears the maintainers of this project haven't been active in several months.  I've published a version of the extension with the required fixes here.  https://github.com/maddogjt/vscode-arduino/releases/download/v0.3.2%2Bserialfix/vscode-arduino-0.3.2+serialfix.vsix

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

simonl65, 2020-10-29:
@robotdad I've not tried the prerelease build yet but just wanted to say - on behalf of all the plugin users - THANK YOU for your hard work.

simonl65, 2020-10-29:
@robotdad Excellent - it works for me :-)

1. Update VSC to 0.50.1
2. Download 0.3.3-rc
3. Drag the vsix file over the extensions panel to install

Thank you

robotdad, 2020-10-30:
Resolved in the 0.3.3 release

closed 2020-10-30 by robotdad

dude-dou, 2020-10-30:
Thank y'all.