microsoft/vscode-arduino #1120: Serial monitor does not work in VS Code 1.50 due to incorrect Electron version

daggilli, 2020-10-28:
The serial monitor is broken under VS Code 1.50.1 as it requires a `NODE_MODULE_VERSION` (80) that corresponds to electron v9.9.3. There is no mainline `NODE_MODULE_VERSION` of 80. I eventually fixed this problem after much experimentation by cloning the module, building it, and then using electron-rebuild with `-v 9.9.3` . I ran the gulp install script and the postinstall process and copied the generated `out` directory to the extensions path for Code.

This was time-consuming and tiresome. I should not have had to do it. Surely there is a regression suite that can be run against VS Code.

robotdad, 2020-10-29:
Prerelease build for the serial port issue is here: https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.3-rc

I plan to publish a release with this fix tomorrow, Oct 30.

boskoe, 2020-10-29:
0.3.3-rc resolves serial port problems for me.

Version: 1.50.1 (system setup)
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T15:06:15.712Z (2 wks ago)
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

boskoe, 2020-10-29:
I spoke too soon. 0.3.3-rc does allow me to select COM ports and open/close said port. However no serial output comes from an open connection. If I close the COM port connection and connect with Putty I seen the output coming from the board.

robotdad, 2020-10-29:
Thank you for opening a separate issue on the serial monitor. I'd prefer to get this release out so deploy works again while investigating #1125

robotdad, 2020-10-30:
Resolved in the 0.3.3 release

boskoe, 2020-10-30:
I'm sorry but 0.3.3 did not resolve the issue of no serial output

robotdad, 2020-10-31:
Thanks, reactivated and tagged against serial for when we push on that next.

boskoe, 2020-11-03:
Any idea on when this push might be?

adiazulay, 2020-12-08:
Are you still having issues with serial port?

boskoe, 2020-12-08:
It seems to be resolved. 

Thanks

closed 2020-12-08 by adiazulay