microsoft/vscode-arduino #1122: Extension issue

love2nerd, 2020-10-28:
- Issue Type: `Bug`
- Extension Name: `vscode-arduino`
- Extension Version: `0.3.2`
- OS Version: `Linux arm 5.4.72-v7l `
- VSCode version: `1.50.1`

:warning: We have written the needed data into your clipboard. Please paste! :w```json 
{
	"messages": [],
	"activationTimes": {
		"codeLoadingTime": 178,
		"activateCallTime": 126,
		"activateResolvedTime": 0,
		"activationReason": {
			"startup": false,
			"extensionId": {
				"value": "vsciot-vscode.vscode-arduino",
				"_lower": "vsciot-vscode.vscode-arduino"
			},
			"activationEvent": "*"
		}
	},
	"runtimeErrors": []
}
```arning:

robotdad, 2020-10-29:
What is the issue?

robotdad, 2020-11-13:
Closing for now, reopen with details on what the issue is e.g. repro steps with the current behavior and expected behavior.

closed 2020-11-13 by robotdad