microsoft/vscode-arduino #1125: 0.3.3-rc allows selection of COM port and connecting but no serial data is read

boskoe, 2020-10-29:
0.3.3-rc **does** allow me to select COM ports and open/close said port. However no serial output comes from an open connection. If I close the COM port connection and connect with Putty I seen the output coming from the board.

Version: 1.50.1 (system setup)
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T15:06:15.712Z (2 wks ago)
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

robotdad, 2020-10-29:
Odd, I see the x to disconnect on my Win10 machine and it seems to work. I'll see how this behaves elsewhere. I'm not going to block the current release on this issue as deploy has been broken for too long here.

boskoe, 2020-10-29:
@robotdad 
I agree with letting the rc go. I'm going to try and revert to earlier versions. Totally removing vscode and reinstalling doesn't seem to fix.

boskoe, 2020-11-20:
This appears to be resolved in #1140

closed 2020-11-20 by boskoe