microsoft/vscode-arduino #1130: 0.3.3 No serial output when extension indicated it's connected.

boskoe, 2020-10-30:
Connect to a COM port connected to a microcontroller that is serial printing once a second. Nothing is shown in the Output screen other than port connection status. Independent monitoring  of the COM port with Putty shows the output from the microcontroller.

Version: 1.50.1 (user setup)
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T15:06:15.712Z (2 wks ago)
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

boskoe, 2020-11-04:
Well this is embarrassing. I opened this and previous issues based on VSCode and the Arduino 0.3.3-rc and 0.3.3 not having any serial output when connected to a controller. When using the Arduion IDE and this controller I got the expected serial output but not from the VSCode & Arduino Extension. Today plugged in a Adafruit Huzzah32 controller and began getting expected serial output with VSCode and Arduino 0.3.3 extension.

 I have two controllers (Arduino Portenta) that exhibit this problem. The serial output from then works on my computer with Arduino IDE and Putty but not VSCode & Arduino 0.3.3 extension. Now I'm not sure this is a VSCode or Extension problem.

adiazulay, 2020-11-12:
@boskoe I wonder if the Portenta is expecting a DTR signal before it starts transmitting data. We merged #1140 yesterday into the develop branch which sends a DTR signal on serial open. Would you be able to try it out and see if that resolves the issue?

boskoe, 2020-11-12:
@adiazulay In and effort to mitigate this I have gone to VS Code Insiders so yes I would like to try anything and report my findings.
Is there a release candidate of the Arduino extension to try?

boskoe, 2020-11-18:
@adiazulay I was unable to test this as I couldn't figure out how to obtain the updated extension.

adiazulay, 2020-11-18:
@boskoe sorry for the delay, here is the rc with fix in it 
https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.4-rc

boskoe, 2020-11-18:
@adiazulay 
I've tested an Arduino Portenta here at work with v0.3.4-rc and get no serial output. When I check with Putty I get serial output. I then attached a generic ESP32 board and get serial output with v0.3.4-rc, VSCode and Putty. I have another Portenta at home and will test with that one later.

boskoe, 2020-11-18:
@adiazulay 
The Portenta at home exhibits the same behavior. Everything else works fine. This is no reason to hold up this release in my opinion.

boskoe, 2020-11-20:
@adiazulay 

It appears that 0.3.4-rc2 has resolved the serial problem I reported with the Arduino Portenta.
Thank you.

closed 2020-11-20 by boskoe