microsoft/vscode-arduino #1132: It is good to add input function in the Serial Monitor

Jason0-0, 2020-10-31:
It is fantastic that the serial selection and serial monitor work again after weeks of  waiting. 
However, the serial monitor now can only reading strings from the Arduino boards and we can't writing somethings in the monitor console and send it to the boards. It seems that it is the output window the serial monitor outputs to that limits the function of sending strings to the Arduino via the monitor. So is it possible for us to use the terminal window to call the serial monitor?

NbrTDB, 2020-11-16:
See also #463, there is quite a desire for this.

Jason0-0, 2020-11-16:
> See also #463, there is quite a desire for this.

Thank you ! It really helps me a lot

But I still hope that it can integrate in the monitor window, though the support from MS must be required.

closed 2020-11-16 by Jason0-0