microsoft/vscode-arduino #1134: Serial port open - does not reset Arduino boards

iqapps, 2020-11-02:
Applies to Arduino Extension v0.3.3

Serial port does not reset Arduino board after connecting. Seems like DTS signalling is broken on Open serial port.

adiazulay, 2020-11-22:
this is resolved in v0.3.4

closed 2020-11-22 by adiazulay

alka79, 2020-12-11:
If this feature has to stay, could you make it optional ?

In many situations, you just want to open a com port to read messages WITHOUT reset. Current 0.3.4 systematically resets when you open the com port :-(

In my experience, arduino automatically resets when you physically plug to the serial port and after upload, not when a remote Serial communication software opens a com port.

adiazulay, 2020-12-11:
That shouldn't be too hard to do. Can you open a new issue so we can track the feature request?

alka79, 2020-12-11:
done. https://github.com/microsoft/vscode-arduino/issues/1154