microsoft/vscode-arduino #1136: Error compiling for Adafruit nRF52832

QuteSaltyFish, 2020-11-02:
After I choose the board, and press compile, it tell me the error:

[48765] Failed to execute script __main__
Traceback (most recent call last):
  File "nordicsemi/__main__.py", line 315, in <module>
  File "click/core.py", line 722, in __call__
  File "click/core.py", line 676, in main
  File "click/_unicodefun.py", line 118, in _verify_python3_env
RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment.  Consult http://click.pocoo.org/python3/for mitigation steps.

This system lists a couple of UTF-8 supporting locales that
you can pick from.  The following suitable locales where
discovered: af_ZA.UTF-8, am_ET.UTF-8, be_BY.UTF-8, bg_BG.UTF-8, ca_ES.UTF-8, cs_CZ.UTF-8, da_DK.UTF-8, de_AT.UTF-8, de_CH.UTF-8, de_DE.UTF-8, el_GR.UTF-8, en_AU.UTF-8, en_CA.UTF-8, en_GB.UTF-8, en_IE.UTF-8, en_NZ.UTF-8, en_US.UTF-8, es_ES.UTF-8, et_EE.UTF-8, eu_ES.UTF-8, fi_FI.UTF-8, fr_BE.UTF-8, fr_CA.UTF-8, fr_CH.UTF-8, fr_FR.UTF-8, he_IL.UTF-8, hr_HR.UTF-8, hu_HU.UTF-8, hy_AM.UTF-8, is_IS.UTF-8, it_CH.UTF-8, it_IT.UTF-8, ja_JP.UTF-8, kk_KZ.UTF-8, ko_KR.UTF-8, lt_LT.UTF-8, nl_BE.UTF-8, nl_NL.UTF-8, no_NO.UTF-8, pl_PL.UTF-8, pt_BR.UTF-8, pt_PT.UTF-8, ro_RO.UTF-8, ru_RU.UTF-8, sk_SK.UTF-8, sl_SI.UTF-8, sr_YU.UTF-8, sv_SE.UTF-8, tr_TR.UTF-8, uk_UA.UTF-8, zh_CN.UTF-8, zh_HK.UTF-8, zh_TW.UTF-8
exit status 255
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
[Error] Exit with code=1

does anyone have the same issue?

somebody-vip, 2021-04-15:
I also reproduced this issue programming the Adafruit ItsyBitsy nRF52840 Express board on the Arduino IDE 2.0 Beta. But on Arduino IDE Release 1.8.13, it works fine.

ajs123, 2021-05-25:
No solution, but maybe hints...

The following build script works:

{
	"version": "2.0.0",
	"tasks": [
		{
			"type": "shell",
			"label": "arduino-cli build active file",
			"command": "/usr/local/bin/arduino-cli compile --fqbn adafruit:nrf52:feather52840 --build-cache-path /tmp",
			"problemMatcher": [],
			"group": {
				"kind": "build",
				"isDefault": true
			},
			"isBackground": false,
			"presentation": {
				"echo": true,
				"reveal": "always",
				"focus": false,
				"panel": "shared",
				"showReuseMessage": true
			}
		}
	]
}

But the plugin insists upon directly running one of the binaries at /User/username/Libraries/Arduino15/packages/adafruit/tools.

ajs123, 2021-05-26:
This appears to be fixed with version 0.22.1 of the Adafruit nRF52 board package.

jorg-rodsjo, 2022-11-30:
@ajs123 Did you get it working? I'm running the 1.2.0-version of the Adafruit package and Arduino IDE 2.0.2 and I'm getting the same message.