microsoft/vscode-arduino #1146: Add arduino.output setting

Pararera, 2020-11-28:
Hi,
please, add option to add output setting to global settings so we can defind default output location for all projects/sketches.

threeal, 2021-01-20:
It's said in the README that

> Arduino build output path. If not set, Arduino will create a new temporary output folder each time, which means it cannot reuse the intermediate result of the previous build leading to long verify/upload time, so it is recommended to set the field. **Arduino requires that the output path should not be the workspace itself or in a subfolder of the workspace**, otherwise, it may not work correctly. By default, this option is not set. It's worth noting that the contents of this file could be deleted during the build process, so pick (or create) a directory that will not store files you want to keep.

It seems that we need to correct that problem first, as usually the output/build location for other build system laid inside the workspace. Choosing the output folder inside the workspace would be useless and choosing it outside the workspace would be obvious.

Also i think it may correlate with https://github.com/microsoft/vscode-arduino/issues/1103

Pararera, 2021-01-20:
> It's said in the README that
>
>
> It seems that we need to correct that problem first, as usually the output/build location for other build system laid inside the workspace. Choosing the output folder [...]

I guess You did not understand what I mean.
I suggested for (default) global setting for build output folder - eg. `..\output`
So every project/workspace will have `../output` build output path **relative to project/workspace**.
For now, I have to manually enter output path for every project/workspace.

Overwatching, 2021-05-14:
I came here for this EXACT thing.
I have to go through ~200 sketches to make sure they run. (student assignments)
Since there is no global default (even if we have to manually add it in the .json) I have to MANUALLY add it to every file.
It was literally faster for me to write a script that pre-generated a .vscode folder with an arduino.json file in it with the correct file information.

Why can't we have a global default? Even if we have to click a box the first time it runs on a workspace?

Pararera, 2021-05-17:
@adiazulay what happend with this feature request?