microsoft/vscode-arduino #1147: Upload and Serial Monitor management : auto close/reopen

alka79, 2020-11-29:
Hi,
In Arduino IDE, if you upload while Serial port is open, the IDE closes temporarilly the com port and re-opens it automatically at the end of the upload. The board is reset _once_ at the end of the upload.

In VSCode-Arduino I must manually close the Serial Monitor prior to upload, and click again to open it when the upload is done. 
The upload resets the board. Re-open the Serial resets the board once again. That's _two_ resets.

Interestingly, when I upload while the Serial Monitor is open, VSCode closes the com port and re-opens it immediately. This happens at the beginning of the compilation. Thus, the upload fails because com port is busy. 
I see the intention to manage properly the Serial Monitor during uploads but it seems to fail. 

Do I miss something ?

I run VSCode 1511,  arduino extension 0.34 with  NodeMCU ESP8266 boards.

alka79, 2020-12-09:
Am I the only one with this issue ? something must be wrong in my setup if no one else complains. It is quite painful in everyday use.

As of now, when I click upload and Serial is open ( a common situation in dev/debug) : 
- Serial is closed
- Serial is immediately reopened (which in 0.34 leads to a reset of the board)
- compile
- if compile success, upload fails because Serial is busy :(
- Serial remains closed. When I reopen it with the button in 0.34, the board is reset once again :(

.
Expected behavior when user clicks Upload : 
- compile
- if compile fails -> exit without touching the Serial port.
- after compile success, if Serial port is open, temporarily close it _(with Arduino IDE, this is when the Serial window remains open but is greyed out)_
- upload
- after upload (successful or failed), if Serial was temporarily closed, reopen it.
- if upload was successful, reset the board only once

nbaztec, 2020-12-20:
I would appreciate the feature as well

closed 2023-04-02 by github-actions[bot]