microsoft/vscode-arduino #1148: Setup the include paths automatically

Stig124, 2020-11-29:
I have a recurring issue with the VSCode Arduino ext, I work with many libraires and every single one breaks the spellchecking (include issues), I managed to make a workaround but it's still a pain

It would be nice if the extension could prefill the include paths automatically

adiazulay, 2020-11-30:
@Stig124  Which version of the extension are you using? v0.3.4 added a fix to intelliSense that resolves most issues, you might need to recreate the c_cpp_properties.json file. To do that open the .vscode folder and delete the file, then open the board manager and reselect the board you're using.

#1141 should fix all the issues once we merge it in.

alka79, 2020-12-10:
> 
> 
> v0.3.4 added a fix to intelliSense that resolves most issues, 

really ?

I currently work on an ESP8266 ESP12E project. It has multiple .ino files, uses libraries from the core and local libs. The main sketch file myproject.ino holds the globals declarations and functions prototypes. Functions declarations are in separated files.

It compiles fine with both Arduino IDE 1.8.1.3 (latest) and VSCode Arduino 0.3.4.

The brand new c_cpp_properties.json file, generated by 0.3.4,  is like this : 
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\Al\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "D:\\Al\\Documents\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                "C:\\Users\\Al\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\**"
            ],
            "forcedInclude": [  ],
            "defines": [
                "USBCON"
            ],
            "intelliSenseMode": "msvc-x64",
            "cStandard": "c17",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}
```
I had to add "ARDUINO=10813" in the defines because many libraries check on the Arduino version.
10813 is the value for the latest v1.8.1.3. I think it should be added by default, but it is not the main subject of this comment.

With these setting, VSCode arduino 0.3.4 reports **340** errors in my code. All are false positives.

Then I added this line to forcedInclude : (see my comment https://github.com/microsoft/vscode-arduino/issues/271#issuecomment-727564776)
```
"C:\\Users\\Al\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\cores\\esp8266\\Arduino.h"
```
This should be useless, because this path is already in the include path. But it is not !
With this added line, the number of false positive goes down to **252**

Then I added a second line to forcedInclude
```
"${workspaceFolder}\\myproject.ino"
```
Again, this line should be useless, but it is not. With this, the number of false positive errors goes down to **31**.

That is still a lot!

Here are examples of the remaining 31 false positive reported. 
These are so strange that I can't believe it can be fixed in c_cpp_properties.json.

Example 1: configTime
[image]
This is standard code for ESP8266. MTZ is properly defined.
Seems that configTime() is not found by Intellisense with these input types. But it is found if I navigate to definition and found by the compiler.
IntelliSense seems lost after that line and reports many false errors.

Example 2: 
[image]
On the first line Intellisense says "ID digitalPinToInterrupt not defined". 
Same function, same arguments type, first reported as an error second not. Navigation finds each correctly.

Example 3: nullptr
[image]
This in my main loop(). Intellisense says "too many arguments in function call". 
In another file shown in example 1, the same instruction is correctly recognized. Why not here ?

adiazulay, 2020-12-10:
@alka79 I think what I should have said was is that some issues were resolved in v0.3.4, but we're still working on a more permanent solution.

I should be able to get a fix out for the first two issues you have, I'll look into the last one and see what I can do.

alka79, 2020-12-10:
looking forward to it :) 

Is there a precedence order in the paths listed in includePath and forcedInclude ?

adiazulay, 2020-12-16:
Tracking this with #438

closed 2020-12-16 by adiazulay