microsoft/vscode-arduino #1150: Serial port support in 0.3.4 is broken again in VSCode 1.52.0 Insiders

boskoe, 2020-11-30:
I realize this is new about the Insiders build but it's the same old problem with the serial port support in the Arduino extension.

There  must be a way to keep this extension up to date with VSCode.

Today I started getting this error.

`Failed to open serial port COM15 due to error: + Error: The module '\\?\c:\Users\user\.vscode-insiders\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 80. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).`

This is on Insiders version:

`Version: 1.52.0-insider (user setup)
Commit: 0a80aacc7be1ab03ec0f94b8ac1a84949a83f35d
Date: 2020-11-26T07:36:22.965Z
Electron: 11.0.2
Chrome: 87.0.4280.63
Node.js: 12.18.3
V8: 8.7.220.24-electron.0
OS: Windows_NT x64 10.0.19041`

adiazulay, 2020-11-30:
Thanks for catching this!

boskoe, 2020-12-02:
@adiazulay 

This was resolved in todays Insider update:

Version: 1.52.0-insider (user setup)
Commit: bf21395291146c014220eb70b54fdfc0a15956b5
Date: 2020-12-01T09:19:32.000Z
Electron: 9.3.5
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19041

closed 2020-12-02 by boskoe