microsoft/vscode-arduino #1155: Same code, does not work when compiled using vscode-arduino

edolis, 2020-12-13:
Hi folks,
I have a simple sketch to connect an ESP to a MQTT broker (mosquitto).
The sketch uses 
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
Well, I don't know what' wrong but when I compile and upload the sketch using this extension, the MQTT connections fails, while when compiling and uploading using the Arduino IDE it works correctly. Looks like the MQTT broker does not even get the requests.
The only warning I get from intellisense is linked to a global variable declared in a h file, which seems not to be processed because uint8_t type seems to be undefined, and I have not yet understood how to fix it.
I'm afraid if I cannot understand if the same code works as good as it is compiled with Arduino IDE, it's really hard using this extension for development.
Please suggest if I am missing something here... spent hours trying to make it work, but I suspect there might be an issue with the compiler and I don't know how to fix it.

edolis, 2020-12-22:
got it work. see other issue I closed - it was a configuration issue of the extension

closed 2020-12-22 by edolis

jacklu333333, 2021-01-14:
Hi, I have the same problem with a different library. Can you share the source of reference?
In my case, the Library is included, but compiling keeps show the class has no reference.
The library is tzapu/WiFiManager (https://github.com/tzapu/WiFiManager), and the board is esp32-pico-d4.
The compiler shows that "WiFiManager" was not declared, which is from the tzapu/WiFiManager.

jacklu333333, 2021-01-14:
> got it work. see other issue I closed - it was a configuration issue of the extension

Hi, I have the same problem with a different library. Can you share the source of reference?
In my case, the Library is included, but the compiler keeps show the class has no reference.
The library is tzapu/WiFiManager (https://github.com/tzapu/WiFiManager), and the board is esp32-pico-d4.
The compiler shows that "WiFiManager" was not declared, which is from the tzapu/WiFiManager.