microsoft/vscode-arduino #1156: README and examples are opened in vscode every time the board is detected

jaredjensen, 2020-12-14:
Whenever the board is detected or whenever I upload a sketch, the README and Arduino Example windows are opened in vscode.  If those windows are already open, duplicate windows are created.  I've tried disabling `arduino.enableUSBDetection`, but that didn't work.

How can I disable this behavior?

adiazulay, 2021-01-04:
Can you add a screenshot of the README and Example that open?

jaredjensen, 2021-01-04:
[image]

I just re-installed the extension v0.3.4.

rohitramu, 2021-03-19:
+1, please add an option to disable this behavior.

rohitramu, 2021-03-19:
In case it helps: I've noticed that it actually increases the number of "Arduino Examples" tabs that open each time I reset the board.  For example, if I open VS Code and then reset the board, 1 "Arduino Examples" tab will open.  After closing that tab, if I reset the board again, 2 "Arduino Examples" tabs will open at the same time.  This number keeps going up each time I reset the board.