microsoft/vscode-arduino #1157: Serial port issues. Insiders edition

boskoe, 2020-12-14:
Open an existing, working, Arduino sketch. Attempt to use any of the serial port functions necessary in the Arduino programming environment.

Version: 1.53.0-insider (user setup)
Commit: 96b426ef1da0f0a51ce9ef1931cd1fd2322547e4
Date: 2020-12-14T05:59:27.649Z
Electron: 11.0.3
Chrome: 87.0.4280.67
Node.js: 12.18.3
V8: 8.7.220.25-electron.0
OS: Windows_NT x64 10.0.19041

```
Failed to open serial port COM10 due to error: + Error: The module '\\?\c:\Users\dougcar\.vscode-insiders\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 80. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

adiazulay, 2020-12-14:
Thanks for reporting this, we'll track this and have a fix for this when it goes in to production.

adams13x13, 2021-02-04:
Well, VS Code 1.53 is out, serial port does not work. :-(

adiazulay, 2021-02-05:
I'm working to resolve this right now. We should have a fix out in the next couple of days.

adiazulay, 2021-02-06:
This is fixed in version v0.3.5 you can download and install it manually from the [here](https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.5). The new version will be uploaded to the marketplace in the next few days.

boskoe, 2021-02-06:
@adiazulay 

I have verified that v0.3.5 is working in the following:

VSCode 1.53.0 (users setup), IS Windows_NT x64
VSCode 1.54.0-insiders (user setup), Windows_NT x64 and Darwin arm64 (Mac M1)

Thank you.

closed 2021-02-06 by boskoe

adams13x13, 2021-02-07:
It works, thank you!

Tested with VS Code 1.53.0 on Linux and manual installation of "vscode-arduino-0.3.5.vsix" over "Extensions (on the left)" -> "..." -> "Install from VSIX"

Version: 1.53.0
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T15:56:04.185Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Linux x64 5.4.0-65-generic