microsoft/vscode-arduino #1163: Cannot select board type

nerdthatnoonelikes, 2020-12-21:
[image]
 Nothing is showing when i try to select the board type but i am able to select the serial port. I am able to use Arduino IDE to upload sketches though.

OS: Arch Linux

Starkiller4011, 2020-12-24:
Same issue as nerdthatnoonelikes though when I tried to set the Board Type through the command pallet I got "No supported board is available.\n\nSource: Arduino (Extension)". OS: Manjaro

nerdthatnoonelikes, 2021-01-04:
So i tested the extension on my machine with Xubuntu 20.04 and the extension worked fine. Is there a reason It worked on Xubuntu and not Arch Linux?

nerdthatnoonelikes, 2021-01-05:
So today I reinstalled Arch Linux and installed the Arduino extension for vscode and I can select the board. But I cannot click on the <Select Serial Port> Button. Its completely unresponsive.

ben-kuhn, 2021-01-30:
Same problem on Ubuntu 20.10

Aumandaris, 2021-04-23:
Same problem on Manjaro KDE Plasma 21.0.2

JorgeMarinC, 2021-04-25:
Same problem here on Ubuntu 20.04

JuanDavidBallesteros, 2021-06-21:
I solved this problem, you need to download the Arduino official version (later version of 1.6). The version has a install.sh file, you can install those files  running in console from the folder the command `sh file_name.sh`

Then, the paths must look something like this, so update the paths:

`"arduino.path": "/home/redes/Descargas/arduino-1.8.15",
  "arduino.commandPath": "arduino"`

Finally, you must to restart vs code

orlade, 2021-08-14:
I also get this behaviour when trying to use `arduino-cli`. It works when not using the CLI.

yuxiaoyuan0406, 2021-09-13:
> I solved this problem (Linux), you need to download the Arduino official version (later version of 1.6). The version has a install.sh file, you can install those files running in console from the folder the command `sh file_name.sh`
> 
> Then, the paths must look something like this, so update the paths:
> 
> `"arduino.path": "/home/redes/Descargas/arduino-1.8.15", "arduino.commandPath": "arduino"`
> 
> Finally, you must to restart vs code

Run `whereis arduino` gives `/usr/local/bin/arduino`.   
So I set my settings like this:

```
    "arduino.path": "/usr/local/bin",
    "arduino.commandPath": "arduino"
```

Still not working.

andrewsno1, 2021-09-13:
I had the same issue: I could not select the board even though I had set `path` and `commandPath`.

As soon as I created a .vscode folder in the workspace of my project and created a file called `arduino.json` it sudenly worked. I could select the boards and the selected board would also appear in the `arduino.json` file.

I hope this helps.

DayLightDancer, 2021-09-30:
I had the same issue but I am on windows and I solve it in that way:
VSCode uses the arduino app the get the board info so:
1. Open the Arduino app and setup the board
2. Close the Arduino app - it saves changes on close
3. Open VSCode set the propper path in the fields
4. Click outside of the edited field( sometimes VSCde lost field data if the tab is closed while the edited field is in focus )
5. Restart the VSCode

plejsq, 2021-10-04:
F1 and after that Arduino:initialize have solved problem. Thank you @rrakso

Shaladdin, 2021-10-06:
**I found a Solution** (in windows 10, [but maybe work for you too?])
I download the Arduino(official) version 1.8.13
and change the Arduino extension in vsCode to version v0.3.4

and it works!
now I can select board!

YvanLin, 2021-10-09:
> Downgrading to v0.4.5 fixed this for me (on MacOS 10.15.7 Intel)

It does help! I guess there are some bugs in v0.4.6.

magouero, 2021-10-09:
same here.
reverted arduino extension from 0.4.6 to 0.4.5 and Arduino board configuration comes back to life.
___________________
vscode about:
Version : 1.61.0 (system setup)
Commit : ee8c7def80afc00dd6e593ef12f37756d8f504ea
Date : 2021-10-07T18:13:09.652Z
Electron : 13.5.1
Chrome : 91.0.4472.164
Node.js : 14.16.0
V8 : 9.1.269.39-electron.0
OS : Windows_NT x64 10.0.19043

gustafwigren, 2021-10-13:
Thanx, Reverting extension from 0.4.6 to 0.4.5 also fixed this for me on windows10

AlexIII, 2021-10-31:
Can confirm reverting from 0.4.6 to 0.4.5 fixes the problem (win10)

han-fastolfe, 2021-10-31:
How do you get v0.4.5? I can only see v0.4.6 to install

MysteryMS, 2021-10-31:
> How do you get v0.4.5? I can only see v0.4.6 to install

You should be able to select a specific version by clicking on "Install Another Version" from the Uninstall dropdown.
[image]

Edwardhk, 2021-11-01:
Reverting to version 0.4.5 solved the problem too. (Ubuntu 20.04)

benmcmorran, 2021-11-06:
Fixed in v0.4.7.

closed 2021-11-06 by benmcmorran

GShashikanth, 2021-11-30:
I spent hours on setting Vs code for arduino, but no bingo, can anyone post step by step working process.

RanzQ, 2021-12-18:
If you installed arduino using snap:
```sh
$ which arduino
/snap/bin/arduino
```
I figured out that the paths are a bit weird. But these seem to work:
```json
"arduino.path": "/snap/arduino/current",
"arduino.commandPath": "../../bin/arduino"
```
Without that relative command path, the snap's `/snap/arduino/current/bin/java` isn't found.

YvanLin, 2021-12-18:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

laestrada, 2021-12-20:
Was able to get this working on MacOS Monterey 12.1 with `arduino-cli` and vscode extension 0.4.8. Some initialization was needed using the arduino cli:
1. `$ arduino-cli config init` ( this step may not be necessary )
2. `$ arduino-cli board list` to get the core version
3. `$ arduino-cli core install <insert-core-version-here>`
4. Restart vscode

zakkhoyt, 2022-01-05:
`arduino-cli board list` failed for me on macOS 12.1, but I was able to get running with `arduino-cli core install adruino:avr`

notblank00, 2022-03-28:
reverting did not fix anything on ubuntu 21.10

Mek101, 2022-04-20:
Re open, this issue is still present

crashbit666, 2022-04-27:
Yes, this  issue is still present. I've install arduino-ide form snap. I've configure arduino paths and it working well. But when I try to select arduino board, empty list is showed. I'm using latest version of plugin and latest version of ide on ubuntu 22.04

a1usha, 2023-01-03:
@laestrada Thanks, was missing arduino:avr core. After `arduino-cli core install <insert-core-version-here>` everything initialised properly. Plugin version 0.4.12

PR #1367 (2021-11-05): Fix empty board and examples list
This project is still running on react 15, and these `UNSAFE_*`-lifecycle aliases where not introduced until 16.3, hence they are never called.

fixes #1163, fixes #1360, fixes #1363, fixes #1361, fixes #1372