microsoft/vscode-arduino #1165: #define cli() __asm__ __volatile__ ("cli" ::: "memory")

fredOnGitHub, 2020-12-25:
Hello, I have a problem to recognize interruptions with UNO Rev 3

I get 

```
#define cli() __asm__ __volatile__ ("cli" ::: "memory")
Se développe sur :

__asm__ __volatile__ ("cli" ::: "memory")
identificateur "__asm__" non définiC/C++(20)
Peek Problem (Alt+F8)
No quick fixes available
```

My JSON is
```
...
 "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE",
                "__AVR_ATmega328P__",
                "USBCON"
            ],
...
```

Ok for Serial with the add of "USBCON" but for
**cli()**
**TCCR2A**

fredOnGitHub, 2020-12-31:
All is OK now because :
- all my files are .ino
- the main .ino contains all STRUCT and variables
- others .ino are only for functions (like classes)

The time before with .h I got : 
- a lot of errors and warnings
- modify the json from Arduino (without success)
- #include, .., #ifndef _A_H_, etc .

adiazulay, 2021-01-05:
I think #1141 should resolve this

fredOnGitHub, 2021-01-10:
Hello,

i've just read the message. I 'll watch about it on fewer minutes...

fredOnGitHub, 2021-01-10:
Hello,

i have reinstalled vs code plugin from vs code but it ithe problem still get.

I don't know how to use   [#1141](https://github.com/microsoft/vscode-arduino/pull/1141)

itopaloglu83, 2021-03-05:
Majority of the library configuration is done via the help of preprocessor macros. These macros depend on certain directives to be provided at compile time. For example microcontroller name `__AVR_ATmega328P__ `

These directives are later used in the library to either turn on or off certain functionalities. And that's exactly where your code is failing. Avr-gcc has an extensive list of command line arguments that it later converts to macro definitions. You may see the full list at the address below.

https://www.nongnu.org/avr-libc/user-manual/using_tools.html

In your case including the chip type in `c_cpp_properties.json` should resolve all the issues. Here's an example from a project I have. All you need to do is to add the `__AVR_ATmega328P__` to the defines array. 

```
{
    "configurations": [
        {
            "name": "Mac",
            "includePath": [
                "/Applications/Arduino.app/Contents/Java/tools/**",
                "/Users/<your_username_here>/Downloads/Developer/Arduino/Sketches/libraries/**",
                "/Applications/Arduino.app/Contents/Java/libraries/**",
                "/Applications/Arduino.app/Contents/Java/hardware/tools/**",
                "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/**"
            ],
            "forcedInclude": [
                "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h"
            ],
            "defines": [
                "USBCON",
                "__AVR_ATmega328P__"
            ],
            "macFrameworkPath": [
                "/System/Library/Frameworks",
                "/Library/Frameworks"
            ],
            "intelliSenseMode": "macos-clang-x64",
            "compilerPath": "/usr/bin/clang",
            "cStandard": "c11",
            "cppStandard": "c++98"
        }
    ],
    "version": 4
}
```

You may find more information here https://github.com/microsoft/vscode-arduino/issues/438#issuecomment-791163149 about why it happens.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay