microsoft/vscode-arduino #1166: Cannot find the sketch file

OpenSourceDevelop, 2020-12-28:
"Cannot find the sketch file" error when verifying/uploading code #427

hello, i have the same problem with Windows also Debian... would like getting it to work..

adiazulay, 2021-01-05:
Is the sketch in a folder with the same name?

OpenSourceDevelop, 2021-01-06:
I tried it anyway, but I'm not sure whether the file or the path are correct. can you specify the path and the content of the file?

AtCra, 2021-01-14:
> I tried it anyway, but I'm not sure whether the file or the path are correct. can you specify the path and the content of the file?

I meet the same problem , and it had been solved.
I found the possible reason is you open multiply directions and the ".vscode/arduino.json" **not exist in your project root direction**
This is my test project directory structure:
[image]

OpenSourceDevelop, 2021-01-17:
Path: /home/XXX/Downloads/arduino-1.8.13/

/home/XXX/.vscode/arduino.json

{
"sketch":"GetStarted.ino",
"port":"COM6",
"board":"GetStarted:LALA",
"configuration":"upload_method=OpenOCDMethod"
}

/home/XXX/.vscode/GetStarted.ino

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

**any Idea?**

AtCra, 2021-01-17:
> Path: /home/XXX/Downloads/arduino-1.8.13/
> 
> /home/XXX/.vscode/arduino.json
> 
> {
> "sketch":"GetStarted.ino",
> "port":"COM6",
> "board":"GetStarted:LALA",
> "configuration":"upload_method=OpenOCDMethod"
> }
> 
> /home/XXX/.vscode/GetStarted.ino
> 
> void setup() {
> // put your setup code here, to run once:
> }
> 
> void loop() {
> // put your main code here, to run repeatedly:
> }
> 
> **any Idea?** error "cannot find sketch file"

move
/home/XXX/.vscode/GetStarted.ino
to
/home/XXX/GetStarted.ino?
My ino file not in the  .vscode dir

OpenSourceDevelop, 2021-01-17:
Thank you. But the same Problem "cannot find sketch file"

also in /home/XXX/.vscode/GetStarted.ino
also in /home/XXX/.vscode/GetStarted/GetStarted.ino
also in /home/XXX/Downloads/arduino-1.8.13/GetStarted/GetStarted.ino
also in /home/XXX/Downloads/arduino-1.8.13/GetStarted.ino

:(

is the installation path right? is the same like the Arduino IDE which is working.

adiazulay, 2021-01-25:
do you have Arduino installed in the Downloads folder? If it's not installed in the default location you'll have to point the extension to where Arduino is installed
[image]

then to open the folder containing the sketch go to **File>Open Folder** and select the Get Started directory

closed 2021-02-07 by github-actions[bot]