microsoft/vscode-arduino #1169: [Error] Exit with code=1

Checho05, 2020-12-31:
I was programing with Vscode and the Arduino Extension and when I try to upload the code or to verify the code it sends me this Error :[Error] Exit with code=1. I cant find any fix pleas help.

adiazulay, 2021-01-04:
Does the same sketch verify and upload using the Arduino IDE?

cdgriffin, 2021-01-06:
I had this problem and fixed by opening the sketch (ino) file and on the status bar at the bottom of the VS Code window: selected/changed Language Mode from arduino-output to C++ before uploading.  Check the other options (Sketch File, Show Board Config, Select Serial Port) are correct.

790620263, 2021-01-14:
> Does the same sketch verify and upload using the Arduino IDE?

I had this problem,too.
[image]
And if I use arduino ide directly , it works well... Both compile and upload have no error

cdgriffin, 2021-01-14:
When I get the "Exit with code=1 error I check the c_cpp_properties.json  and arduino.json configurations in the .vscode folder in the same folder as my program ino file.  I also make sure the selections in the bottom status bar are right.  If not I click on them to change them.
[image]
[image]
[image]

AtCra, 2021-01-14:
> When I get the "Exit with code=1 error .......
Thanks for your advice! 
I had solved this problem,and the main reason is PATH ERROR.
.vscode dir placed in the another path, not arduino project root path.
Now it works well.

AtCra, 2021-01-14:
> When I get the "Exit with code=1 error I check the c_cpp_properties.json and arduino.json configurations in the .vscode folder in the same folder as my program ino file. I also make sure the [...]

Thanks for your advice!
I had solved this problem,and the main reason is PATH ERROR.
.vscode dir placed in the another path, not arduino project root path.
Now it works well.

FuryMartin, 2021-09-14:
I met this problem today and have solved it successfully.
The problem might be that there are some non-ascii characters (maybe Chinese, Japanese or other language characters) in the path of your project. For me, I have some Chinese characters in my project path and when I use path without Chinese characters, the extension works well. 

我今天也遇到了这个问题并且顺利的解决了它们，我的情况和上面的不太一样。
问题可能出在你的项目路径上，你的路径里不应该包含中文。我今天的路径里存在中文，当我把文件夹改成英文名之后，插件正常运行了。

heedaelee, 2021-11-10:
in my case, press f1 > arduino: Board Config

there, upload Speed setting was wrong.  That solved my problem 

[image]

hemantshelar, 2022-03-30:
Is anyone facing this problem while configuring VS Code for MXCHIP development? 

I have created a simple IoT Project using [Azure IoT Device Workbench](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-iot-workbench) extension.

After loading the default configuration I can see the following error on the output window.

```
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
exec: "/bin/arm-none-eabi-g++": file does not exist
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch 'device.ino': Exit with code=1
```
I found that, project has a file named `cpp_properties.json`. By default compilerPath is pointing to `C:\\Users\\heman\\AppData\\Local\\Arduino15\\packages\\AZ3166\\tools\\arm-none-eabi-gcc\\5_4-2016q3\\bin\\arm-none-eabi-g++`  but `tools` folder is empty

Any idea how can I get this tool installed on my Windows 10 dev machine?

Here is the [original stack overflow question](https://stackoverflow.com/questions/71670446/5-4-2016q3-bin-arm-none-eabi-g-not-found-windows-10).

Avesta03, 2023-05-07:
[image]

Still getting errors myself - is there an issue here perchance?

tauzn-clock, 2023-06-20:
I am using PlatformIO and it seems to be able to by pass the issue

FMVSC, 2024-02-02:
I was able to fix this issue by editing my include commands for libraries to be in the form of "#include "library.h";" instead of using the <> signs around the "library.h" file. Also, in the MacOS version of VScode to my knowledge there is no auto update for the c_cpp_properties folder, so you must manually find and update that folder with the file path to the source code of the library.