microsoft/vscode-arduino #1170: arduino.path

Dsperad0, 2021-01-02:
I don't know what i should fill in the box that says "Arduino: Command Path" whenever i get the arduino.path error. I have the arduino IDE and an arduino and stuff but it wont let me do anything.

adiazulay, 2021-01-05:
"Arduino: Command Path" will default to your Arduino installation if it's left blank. Did you install the Arduino IDE in somewhere other than the default location?

Can you post your OS, Arduino IDE version, VS Code version, and extension version?

xingyunjohn1, 2021-01-16:
> "Arduino: Command Path" will default to your Arduino installation if it's left blank. Did you install the Arduino IDE in somewhere other than the default location?
> 
> Can you post your OS, Arduino IDE version, VS Code version, and extension version?

I have the same question. But please focus on **Windows Store Version**.
Note that folder C:\Program Files\WindowsApps is not supported to be explored by windows default explorer.

`windows 10.0.18363.1256`
`Arduino windows store 1.8.42.0` *downloaded from Microsoft store*
`Vs Code 1.52.1`
`vsciot-vscode.vscode-arduino v0.3.4`

Regards.

adiazulay, 2021-01-22:
The Windows Store Version does not work with the extension. I would recommend uninstalling the the store version and downloading the Arduino IDE from [here](https://www.arduino.cc/en/software).

FilipRDDG, 2021-02-05:
Hi, I'm experiencing issue with arduino.path as well. I had windows app version of Arduino IDE, but I uninstalled it and then installed normally from .exe installer and I now have version 1.8.13. I'm using VSCode version 1.53.0. And arduino extension v0.3.4. 
My arduino folder is in 'C:\Program Files (x86)\Arduino' so I tried this path (both with and without double slashes) and also linux path (/mnt/c/Program\ Files\ (x86)/Arduino) since I'm using WSL as default VSCode Terminal. But none of it worked... I also tried to delete arduino.path line from settings.json (i found on some forum that it can locat it from default settings) but it didn't work as well (I restarted VSCode after each trie and I even restarted computer sometimes but with no success). 
So I have no idea what can be the problem and how to solve it.

xingyunjohn1, 2021-02-05:
For IDE version, path `C:\Program Files\WindowsApps\Arduino` seems usable. 
I found it from the address of reference pages(in broswer).
**But I didn`t explore further after I saw @adiazulay 's reply.**

utamadonny, 2021-04-18:
I'm using Ubuntu 20.04. 
Visual Studio Code 1.55.2

when clicking "select programmer" or "select board type" 
Error Pop Up 
"Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change."

But in my current json setting was :
"arduino.path": "/home/donny/Downloads/arduino-1.8.13"
and indeed my arduino path is /home/donny/Downloads/arduino-1.8.13 

Select and open serial monitor is fine.

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE and Arduino CLI are currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

closed 2023-04-02 by github-actions[bot]