microsoft/vscode-arduino #1175: Serial Port Custom baud rate

i3130002, 2021-01-08:
I've been using Arduino extension for VSCode for a while and I still can't use it's serial port using custom baud rate (Or even something that it has been configured in board manager :D 
Please consider **adding more baud rates** and also giving us the option to **use a custom one**(Currently it do not accept any custom one).
[image]
[image]

kickshawprogrammer, 2021-04-10:
You can change it, at least on windows, but it's a manual process right now.

Open the file .vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\src\serialmonitor\serialMonitor.js in the user's directory.

Go to the bottom of the file and there you will see a SerialMonitor.DEFAULT_BAUD_RATE = 
Set it to the desired value and save.

Restart VSC.

Default speed changed.

I set my default to 74880 since ESP's default to that value...

It would be nice, however, to have this as a JSON setting in the global / project settings to be sure though...

Regards,

i3130002, 2021-04-11:
> You can change it, at least on windows, but it's a manual process right now.

Thanks @kickshawprogrammer  Will check it out.

gcampbell-msft, 2023-02-16:
@i3130002 We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue or if the issue still persists!

i3130002, 2023-02-21:
> tly updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Glad to hear that, Although I'm not in the industry anymore. 

Hopefully, others can verify this. Thanks for your hard work.

closed 2023-04-02 by github-actions[bot]