microsoft/vscode-arduino #1176: identifier "D0" is undefinedC/C++(20)

i3130002, 2021-01-08:
I'm using `ESP8266 NodeMcu` with `Arduino `and `VS code` and getting `identifier "D0" is undefined C/C++(20)` while it works pretty well after uploading.
# Here is my configs:
## **c_cpp_properties.json:**
```
{
    "configurations": [
        {
            "name": "Arduino",
            "includePath": [
                "${workspaceRoot}",
                "C:\\Users\\[USERNAME]\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "C:\\Users\\[USERNAME]\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.2\\**",
                "C:\\Users\\[USERNAME]\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.2\\variants\\nodemcu"
            ],
            "browse": {
                "path": [
                    "C:\\Program Files (x86)\\Arduino\\hardware",
                    "C:\\Program Files (x86)\\Arduino\\libraries",
                    "${workspaceRoot}"
                ],
                "limitSymbolsToIncludedHeaders": true,
                "databaseFilename": ""
            },
            "forcedInclude": [],
            "intelliSenseMode": "msvc-x64"
        },
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\[USERNAME]\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "C:\\Users\\[USERNAME]\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.2\\**"
            ],
        }
    ],
    "version": 4
}
```

## **settings.json:**
```
{
    "C_Cpp.errorSquiggles": "EnabledIfIncludesResolve",
    "workbench.colorCustomizations": {
        "activityBar.background": "#13341A",
        "titleBar.activeBackground": "#1B4925",
        "titleBar.activeForeground": "#F7FCF8"
    },
    "files.associations": {
        "array": "cpp",
        "deque": "cpp",
        "list": "cpp",
        "string": "cpp",
        "unordered_map": "cpp",
        "vector": "cpp",
        "initializer_list": "cpp"
    },
    "arduino.path": "C:/Program Files (x86)/Arduino",
    "arduino.commandPath": "arduino_debug.exe",
    "arduino.logLevel": "info",
    "arduino.allowPDEFiletype": false,
    "arduino.enableUSBDetection": true,
    "arduino.disableTestingOpen": false,
    "arduino.skipHeaderProvider": false,
    "arduino.additionalUrls": [
        "https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json",
        "http://arduino.esp8266.com/stable/package_esp8266com_index.json"
    ],
    "arduino.defaultBaudRate": 256000,
    // "C_Cpp.intelliSenseEngine": "Tag Parser"
}
```

# **Code:**
```
#include <Arduino.h>
#define LED D0   // Led in NodeMCU at pin GPIO16 (D0).
void setup() {
  pinMode(LED, OUTPUT);    // LED pin as output.
  digitalWrite(LED, LOW); // turn the LED on.
}
```

# Defining port with `static const uint8_t D0   = 16;` creates the following error:
```
mpu6050MicrosOutput:2:22: error: redefinition of 'const uint8_t D0'
 static const uint8_t D0 = 16;
                      ^
In file included from C:\Users\[USERNAME]\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266/Arduino.h:296:0,
                        from C\project\mpu6050MicrosOutput.ino:1:
C:\Users\[USERNAME]\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\variants\nodemcu/pins_arduino.h:40:22: error: 'const uint8_t D0' previously defined here
 static const uint8_t D0   = 16;
                      ^
```

# Using `"C_Cpp.intelliSenseEngine": "Tag Parser"`  turns off IDE errors, Although, makes development vulnerable to mistakes. Any solutions?

edolis, 2021-01-09:
have a look at my issue, to see if my fix works for you too https://github.com/microsoft/vscode-arduino/issues/1178

i3130002, 2021-01-10:
Thanks @edolis , as #1178 it works by force including `pins_Arduino.h` to `c_cpp_properties.json` 
```
...
"includePath": [
...
],
"forcedInclude": [
                                "C:\\Users\\[USERNAME]\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.2\\variants\\d1\\pins_Arduino.h"
            ],
...
```

closed 2021-01-10 by i3130002