microsoft/vscode-arduino #1180: problems with intellisense

Makkio88, 2021-01-10:
hi im new in coding for arduino with vs code.
I'm trying to configure it but i have some problems with intellisense.
I've created a blank project for arduino mkr1010wifi.
Initially all works good but if i try to include a library, such as WiFiNINA, intellisense goes mad and starts to tell me that basic core function and classes of the board are unidentified, as in this image
[image]

this is my configuration file
```
"configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.9\\**",
                "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\**",
                "C:\\Users\\Admin\\Documents\\Arduino\\libraries\\WiFiNINA\\src\\**"
                
            ],
            "forcedInclude": [
                "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.9\\cores\\arduino\\Arduino.h"               
            ],
            "defines": [
                "USBCON"
            ],
            "intelliSenseMode": "msvc-x64",
            "cStandard": "c17",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
```

im going mad around this problem...where is the solution?

edolis, 2021-01-10:
(try have a look at that forcedinclude and see if commenting it out helps)

Makkio88, 2021-01-12:
Thank you. I partially solved the problem modifying the "includePath" section.
Now intellisense works with core functions and imported libraries, but i have an error with #include directives.

adiazulay, 2021-01-22:
This issue is being tracked with #438 and is fixed in #1183. There will be a release candidate with the fixes in the next week or so.

closed 2021-01-22 by adiazulay