microsoft/vscode-arduino #1181: usb-detection module not available for arm32 devices ("wrong ELF 64 class" error)

lewixlabs, 2021-01-10:
Hi and happy new year! :)
My workstation is
- Raspberry Pi4 8GB with Raspi OS v10 for arm32.
- VsCode v1.52.1 for arm32

My experience:
- Arduino Ide for arm32 works fine with my Elegoo (Arduino compatible) Uno R3
- Vs Code Arduino Plugin works but Serial Monitor option crashes with error "wrong ELF 64 class" for usb-detection module

As I suspected I thinks it's a architecture issue, so I tried to build from scratch this extension, following @kseyan guide (I think "npm i -g gulp" command is not necessary) and extensions works fine with Serial Monitor option.
This is the link for anybody would need it:
https://github.com/microsoft/vscode-arduino/issues/1113#issuecomment-713100975

Maybe isn't there a full arm32 compatible plugin version?

Lewix

egandro, 2021-02-22:
This is related to this issue: https://github.com/microsoft/vscode-arduino/issues/1207

gcampbell-msft, 2023-02-16:
@lewixlabs We have recently updated the Arduino extension to depend on a new standalone Serial Monitor extension, please upgrade your arduino extension and try out this new feature!

Let us know if it solves your issue or if the issue still persists!

lewixlabs, 2023-02-17:
Thank you @gcampbell-msft for reporting me this update.
I' sorry this is an issue of 2 years ago and today I'm using Raspberry Pi OS arm64, a different architecture.
But thank you anyway, this update can help other devs in arm32 👍

closed 2023-02-17 by gcampbell-msft