microsoft/vscode-arduino #1182: vscode-errors when using bluetooth (esp32)

v11, 2021-01-10:
Hello

I am using the vscode-arduino extension now for more than one year. Except with a few issues at beginning, i usually don't have any problem to work with my ESP32 and some other libraries.

Now, i started to work with [bluetooth](https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE).

My code compiles and i could successfully test it. But: I get a lot of errors in vscode.
See: https://vision11.ch/share/temp/Bildschirmfoto%202021-01-10%20um%2022.10.30.png

This is my code:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_write/BLE_write.ino

Can somebody please help me? I really want to ged rid of this errors.

Thank you,
Lukas

This is my c_cpp_properties.json
```
{
    "configurations": [
        {
            "name": "Mac",
            "includePath": [
                "${workspaceFolder}/**",
                "/Users/vonnielu/Library/Arduino15/packages/esp32/tools/**",
                "/Users/vonnielu/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/**"
            ],
            "forcedInclude": [
                "/Users/vonnielu/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Arduino.h"
            ],
            "macFrameworkPath": [
                "/System/Library/Frameworks",
                "/Library/Frameworks"
            ],
            "intelliSenseMode": "clang-x64",
            "compilerPath": "/usr/bin/clang",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "defines": [
                "USBCON"
            ]
        }
    ],
    "version": 4
}
```

My ardunio.json
```
{
    "board": "esp32:esp32:esp32",
    "configuration": "PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=115200,DebugLevel=none",
    "sketch": "test.ino",
    "port": "/dev/tty.usbserial-0001",
    "programmer": "AVR ISP"
}
```

My settings.json
```
{
    "files.associations": {
        "__functional_base": "cpp",
        "istream": "cpp",
        "locale": "cpp",
        "memory": "cpp",
        "tuple": "cpp",
        "utility": "cpp",
        "functional": "cpp",
        "__bit_reference": "cpp",
        "array": "cpp",
        "bitset": "cpp",
        "chrono": "cpp",
        "deque": "cpp",
        "iterator": "cpp",
        "map": "cpp",
        "string": "cpp",
        "string_view": "cpp",
        "unordered_map": "cpp",
        "vector": "cpp",
        "*.tcc": "cpp"
    }
}
```

v11, 2021-01-11:
I had to include: #include <string>
Now it is working.

adiazulay, 2021-01-22:
This issue is being tracked with #438 and is fixed in #1183. There will be a release candidate with the fixes in the next week or so.

closed 2021-01-22 by adiazulay