microsoft/vscode-arduino #1186: Intellisense / problems list confused by PROGMEM attribute

DavidShoe, 2021-01-17:
const char string_1[] PROGMEM = "Learn";

results in a problem listed:
{
	"resource": "/c:/projects/vscodebug/vscodebug.ino",
	"owner": "C/C++",
	"code": "65",
	"severity": 8,
	"message": "expected a ';'",
	"source": "C/C++",
	"startLineNumber": 1,
	"startColumn": 23,
	"endLineNumber": 1,
	"endColumn": 30
}

Arduino documentation of the attribute can be found at:
https://www.arduino.cc/reference/en/language/variables/utilities/progmem/

Teraskull, 2021-01-27:
Same issue, my code compiles and uploads.

adiazulay, 2021-02-18:
can you try installing the .vsix from: https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.0-rc2

edleno2, 2021-02-25:
Have the same problem but the 0.4.0-rc2 vsix won't install: "Element 'Identity' has invalid value for attribute 'Version'.  

Section from the install log file: 

Microsoft.VisualStudio.ExtensionManager.InvalidExtensionManifestException: Element 'Identity' has invalid value for attribute 'Version' ---> System.FormatException: Input string was not in a correct format.
   at System.Version.VersionResult.SetFailure(ParseFailureKind failure, String argument)
   at System.Version.TryParseComponent(String component, String componentName, VersionResult& result, Int32& parsedComponent)
   at System.Version.TryParseVersion(String version, VersionResult& result)
   at System.Version.Parse(String input)
   at System.Version..ctor(String version)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.<>c.<ExtractManifestData>b__67_1(String s)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.VerifyStringValue[T](String value, ErrorInfo info, Func`2 selector, Boolean isRequired, Int32 maxLength, T defaultValue)
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.VisualStudio.Telemetry.WindowsErrorReporting.WatsonReport.GetClrWatsonExceptionInfo(Exception exceptionObject)
   --- End of inner exception stack trace ---
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.VerifyStringValue[T](String value, ErrorInfo info, Func`2 selector, Boolean isRequired, Int32 maxLength, T defaultValue)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.ExtractManifestData(PackageManifest manifest)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.ReadPackageManifest(XmlReader reader)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionImpl.ReadManifest(Stream manifestStream)
   at Microsoft.VisualStudio.ExtensionManager.InstallableExtensionImpl.ReadVSIXManifestFromPackage(Stream stream, CultureInfo preferredCulture)
   at Microsoft.VisualStudio.ExtensionManager.InstallableExtensionImpl..ctor(String path, CultureInfo preferredCulture)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionEngineImpl.CreateInstallableExtension(String extensionPath, CultureInfo preferredCulture)
   at Microsoft.VisualStudio.ExtensionManager.ExtensionEngineImpl.CreateInstallableExtension(String extensionPath)
   at VSIXInstaller.ExtensionService.GetInstallableData(String vsixPath, String extensionPackParentName, Boolean isRepairSupported, IStateData stateData, IEnumerable`1& skuData)
   at VSIXInstaller.ExtensionPackService.IsExtensionPack(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.ExtensionPackService.ExpandExtensionPackToInstall(IStateData stateData, Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize(Boolean isRepairSupported)
   at VSIXInstaller.App.Initialize()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()

Same problem with -rc vsix.  3.5 intalls Ok.  

Program does compile under 3.5, but intellisense has problems with PROGMEM attribute.

adiazulay, 2021-02-25:
@edleno2 what version of vs code and what OS are you using?

edleno2, 2021-02-25:
@adiazulay 1.53.2 January according to the notes.  From the VSCode Help/About:
Version: 1.53.2 (user setup)
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:48:04.245Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.18363

Running Windows 10 (above says 10.0.18363).  

I also have Visual Studio 2019 Preview 4.0 installed.  The Visual Studio Installer says version 2.9.3342.

edleno2, 2021-02-25:
@adiazulay Install says vsixinstaller.exe version 16.9.1050.  Install log file attached.
[dd_VSIXInstaller_20210225144342_6bd0.log](https://github.com/microsoft/vscode-arduino/files/6046160/dd_VSIXInstaller_20210225144342_6bd0.log)

edleno2, 2021-02-27:
@adiazulay My bad on the install problem - I wasn't using the VSCode install command within the IDE.  For others: installing from a VSIX is done in VSCode itself - can't just download and let the Visual Studio Installer attempt to do it.  See this link: [Installing from a VSIX](https://code.visualstudio.com/docs/editor/extension-gallery).  Use F1 to get to the command palette and then Extensions: Install from VSIX.

edleno2, 2021-02-27:
@DavidShoe @adiazulay @Teraskull After installing 0.4.0-rc2 I deleted the .vscode folder and reran Arduino:Initialize and it creates an Arduino C++ configuration that goes to the AVR compiler which DOES understand the PROGMEM attribute.  I get a lot of additional configuration with the newer release.  So far I have fewer problems with include paths and for sure the default configuration for the devices is much more robust.  Configurations things that I've had to put in by hand are now done automatically with Arduino:Initialize.

adiazulay, 2021-03-22:
IntelliSense auto-generation has been added in v0.4.0. If you're having any problems with IntelliSense please open a new issue.

closed 2021-03-22 by adiazulay