microsoft/vscode-arduino #1195: SERIAL PORT

NICORUIZ93, 2021-02-05:
Issue Type: <b>Bug</b>

HAY UN PROBLEMA AL SELECCIONAR EL PUERTO SERIAL, NO APARARECE NINGUN PUERTO Y ADEMAS NO PERMITE ABRIR EL MONITOR SERIAL DESDE LA NUEVA VERSION DE VS CODE

Extension version: 0.3.4
VS Code version: Code 1.53.0 (8490d3dde47c57ba65ec40dd192d014fd2113496, 2021-02-03T20:36:38.611Z)
OS version: Windows_NT x64 10.0.19042

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-3610QM CPU @ 2.30GHz (8 x 2295)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.89GB (3.15GB free)|
|Process Argv|--crash-reporter-id 32f319b0-1ea7-432d-8581-afd1f8a11688|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368cf:30146710
vsreu685:30147344
python383:30185418
vspor879:30202332
vspor708:30202333
vspor363:30204092
vswsl492cf:30211402
pythonvsdeb440:30248342
pythonvsded773:30248341
vstes516:30244333
pythonvspyt875:30251590

```

</details>

mathertel, 2021-02-05:
> there is a problem in selecting the serial port, it does not appear in any port and adds does not allow opening the serial monitor from the new version of vs code

not working: 
* Select port
* upload Sketch
* Serial Monitor

adiazulay, 2021-02-05:
See #1157

closed 2021-02-05 by adiazulay