microsoft/vscode-arduino #1197: detection.node was compiled against a different Node.js

JohnKozell, 2021-02-06:
The module '\\?\c:\...\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 80. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

after upgrading to Visual Studio Code 1.53.0

jooftian, 2021-02-06:
I'm having a similar issue: "\detection.node is not a valid Win32 application"

> Failed to open serial port COM3 due to error: + Error: \\?\c:\Users\sebas\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node is not a valid Win32 application

adiazulay, 2021-02-06:
We're tracking this with #1157.

closed 2021-02-06 by adiazulay