microsoft/vscode-arduino #1198: Can't select Arduino serial port (again)

dariocs78, 2021-02-07:
Hello,

Related to previous issues: #922 #1105 #1104 #1101 #1099 #1104

I can't select the Arduino serial port by clicking on the bar button. It works only by editing arduino.json as a workaround.

It seemed this was solved back in version 0.3.3, but it's happening to me again with version 0.3.4 under VS Code 1.53.0 in Windows 10.

Anyone experiencing the same problem?

BR

0xStarcat, 2021-02-07:
i am on `0.3.4` w/ vscode on ubuntu 

```
Failed to open serial port /dev/ttyUSB0 due to error: + Error: The module '/home/lola/.vscode/extensions/vsciot-vscode.vscode-arduino-0.3.4/out/node_modules/usb-detection/build/Release/detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 67. This version of Node.js requires NODE_MODULE_VERSION 85. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).
```

AlphaSierraHotel, 2021-02-07:
Same for me on a Windows platform

`Failed to open serial port COM7 due to error: + Error: The module '\\?\c:\Users\<redacted>\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 80. This version of Node.js requires NODE_MODULE_VERSION 85. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or `npm install`).
`

AlphaSierraHotel, 2021-02-07:
Also, possibly relevant as there was an update applied recently

```
Version: 1.53.0 (user setup)
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T20:36:38.611Z
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19041
```

0xStarcat, 2021-02-07:
Seems like the `NODE_MODULE_VERSION` bug was fixed in `0.3.5` https://github.com/microsoft/vscode-arduino/issues/1074#issuecomment-774477379

so if this is your issue like me, download the `vsix` file from https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.5 

and run `code --install-extension /path/to/.vsix` then reload vscode

emily-curry, 2021-02-07:
> so if this is your issue like me, to fix it I downloaded the `vsix` file from https://github.com/microsoft/vscode-arduino/releases/tag/v0.3.5
> 
> and run `code --install-extension /path/to/.vsix` then reload vscode

Upgrading to extension version `0.3.5` with vscode `1.53.0` (electron `11.2.1`, node `12.18.3`) did the trick for me, thank you!

AlphaSierraHotel, 2021-02-07:
Worked for me too.  Thanks.

moraesjeremias, 2021-02-08:
Worked for me as well, thanks for the advice @0xStarcat !

lamello, 2021-02-08:
Worked for me also. Thank you.

jorgeota, 2021-02-08:
worked for me. Thank you!

adiazulay, 2021-02-08:
Thanks @0xStarcat for explaining how to install the ```vsix```.
I'll leave this issue pinned until the extension publishes to the marketplace.

cdg123, 2021-02-09:
Great thanks for the guidance

hongqiangwang, 2021-02-10:
works for me. thanks a lot!

masynthetic, 2021-02-11:
I swear these never end... see ya'll next month

UID163844189, 2021-02-11:
Worked for me so. Thank you! 
And, you can install the extension in vs code instead of command line.

marcelo-lara, 2021-02-13:
worked for me too =)
Thanks!

rvdbreemen, 2021-02-14:
worked for me to, thanks!

adiazulay, 2021-02-18:
v0.3.5 is now available in the marketplace. Thanks for bearing with us!

closed 2021-02-18 by adiazulay

HowardHonig, 2021-08-07:
Well, it's back again!  VSC v1.59 forced the error **Failed to open serial port ....etc**. Tried upgrading to latest version of the vsix but that didn't fix the problem.

This appears to have happen on most releases of VSC.  Doesn't anyone check this BEFORE releasing the code?  Downgrading to VSC v1.57 fixed the problem but VSC still keeps asking to upgrade.

munef, 2021-08-15:
I faced the same issue with VSC v1.59 and fixed by rolling back to v1.56.2. I disabled auto update by Preferences > Settings (Ctrl + ,) > Update: Mode to "none". Reference: 
https://stackoverflow.com/questions/50162685/how-do-i-disable-vs-code-of-updating-itself

MaxRev-Dev, 2021-08-24:
@HowardHonig  @munef Just encountered the same. Downgrading VSC to 1.57 or 1.56 does not help. For me installing [v0.4.4-rc1](https://github.com/microsoft/vscode-arduino/tree/v0.4.4-rc1) fixed the issue.

HowardHonig, 2021-08-24:
Great, but the Readme on v0.4.4-rc1 doesn't explain how to install this extension update.  The command palette install didn't work and the arduino extension "install another version" doesn't give me the option of selecting anything from the disc.

teymurov27, 2021-08-24:
Same issue here VSC v1.59 and fixed by downgrading to 1.48.2. Command palette for changing .ino file also did not work out for me.

HowardHonig, 2021-08-24:
MaxRev_Dev: Thanks for the suggestion as I learned a lot more about how extensions are applied.  So now I have to figure out how to make a .vsix file out of all the code.  NO THANKS.  I'll wait for it to be released and just not upgrade VSC.

MaxRev-Dev, 2021-08-24:
@HowardHonig You don't need to create package. Look in release section. https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.4-rc1

HowardHonig, 2021-08-24:
MaxRev-Dev:  Totally didn't see that, and got the rc1 installed.  Thanks for the help.