microsoft/vscode-arduino #1200: COM port errors again - really tiresome

iqapps, 2021-02-09:
Issue Type: <b>Bug</b>

With latest version of VS Code and Arduino extension as of today:

Failed to open serial port COM14 due to error: + Error: The module '\\?\c:\Users\hl\.vscode\extensions\vsciot-vscode.vscode-arduino-0.3.4\out\node_modules\usb-detection\build\Release\detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 80. This version of Node.js requires
NODE_MODULE_VERSION 85. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Extension version: 0.3.4
VS Code version: Code 1.53.0 (8490d3dde47c57ba65ec40dd192d014fd2113496, 2021-02-03T20:36:38.611Z)
OS version: Windows_NT x64 10.0.19042

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i9-9900 CPU @ 3.10GHz (16 x 3096)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>protected_video_decode: enabled<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|31.79GB (17.86GB free)|
|Process Argv|--crash-reporter-id 937b32ac-d42d-4d5c-8600-a5390f81789c|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383cf:30185419
pythonvspyt617cf:30253240
vspor879:30202332
vspor708:30202333
vspor363:30204092
vswsl492:30211401
pythonvsdeb440:30248342
pythonvsded773:30248341
pythonvspyt600cf:30251589

```

</details>

iqapps, 2021-02-09:
Solved by downloading Arduino extension V 0.35 and install.

adiazulay, 2021-02-09:
See #1198

closed 2021-02-09 by adiazulay