microsoft/vscode-arduino #1202: Cannot Select Serial Port

Alvin-Chips, 2021-02-15:
Issue Type: <b>Bug</b>

Install the arduino library, then select your serial port, and nothing should happen. 

Extension version: 0.3.4
VS Code version: Code 1.53.2 (622cb03f7e070a9670c94bae1a45d78d7181fbd4, 2021-02-11T11:48:04.245Z)
OS version: Windows_NT x64 10.0.19041

boskoe, 2021-02-15:
You need 0.3.5

Alvin-Chips, 2021-02-15:
> You need 0.3.5

And how do I install this version?

boskoe, 2021-02-15:
Open the extensions module "Ctrl+Shft+X". Click the three horizontal dots at the top "..." and click "Check for Extension Updates"

Alvin-Chips, 2021-02-15:
It says "All extensions are up to date."

boskoe, 2021-02-15:
Try this then.

https://github.com/microsoft/vscode-arduino/issues/1198#issuecomment-774746151

Alvin-Chips, 2021-02-15:
Ok I have read through a lot of it, and I'm wondering how to manually add a serial port by code.

boskoe, 2021-02-16:
[Let me Google that for you...](https://letmegooglethat.com/?q=vscode+arduino+manually+add+com+port)

Alvin-Chips, 2021-02-16:
Ok Thank you!

AronRubin, 2021-02-16:
I have success now with vscode-arduino 0.3.5 . I did have to download that from the releases and "Install from VSIX"

adiazulay, 2021-02-16:
please see #1198

closed 2021-02-16 by adiazulay