microsoft/vscode-arduino #1203: Output strings are garbled in Japanese environment

HiroTamori, 2021-02-15:
version 0.3.5 (this bug occurred at least 0.3.4 too)
on Windows 10 (Build 1909) Japanese edition
[image]
Workaround is comment the line 215 of out/src/common/util.js out;
```
        const child = childProcess.spawn(command, args, options);
        let codepage = "65001";
/*
        if (os.platform() === "win32") {
            try {
                const chcp = childProcess.execSync("chcp.com");
                codepage = chcp.toString().split(":").pop().trim();
            }
            catch (error) {
                outputChannel_1.arduinoChannel.warning(`Defaulting to code page 850 because chcp.com failed.\
                \rEnsure your path includes %SystemRoot%\\system32\r${error.message}`);
                codepage = "850";
            }
        }
*/
        if (outputChannel) {
            child.stdout.on("data", (data) => {
```
(Thanks nori-dev-akg san on the [article](https://qiita.com/nori-dev-akg/items/e0811eb26274910cdd0e))

This code seems to change UTF-8(codepage 65001) strings to Shift-JIS or something even it is not required by vscode.

Tnze, 2021-02-21:
duplicated : #875, #958, #1096

two-pack, 2021-09-03:
PR #1237 fixed this problem on my environment.
It was already closed but not merged :(

Tnze, 2021-09-05:
But why

two-pack, 2021-09-05:
> PR #1237 fixed this problem on my environment.
> It was already closed but not merged :(

I calmed myself down and checked again.

PR #364 in vscode-arduino added converting using chcp.com command on 13 Jul 2017.
Probably that time, incorrect characters, caused Arduino that output ansi codepage, was fixed.

[PR #8289 in arduino/Arudino](https://github.com/arduino/Arduino/pull/8289) added encoding config of compiler log to arduino.l4j.ini. After that, Arduino outputs UTF-8 codepage by default.
Probably that time, incorrect characters, caused Arduino that output UTF-8 codepage, was occured again.

PR #1237 specifies codepage to UTF-8, so correct characters are out.
But now I think better solution is reverting #364 because default log codepage is UTF-8 on latest Arduino. Or #1271.

benmcmorran, 2022-01-13:
This issue should be fixed by PR #1271 which has merged and will be included in the next release of this extension. If you'd like to test the changes before they are released, you can [download and install the build artifact](https://github.com/microsoft/vscode-arduino/suites/4930939531/artifacts/142502560).

closed 2022-01-15 by github-actions[bot]