microsoft/vscode-arduino #1206: When Complier arg is present in c_cpp_properties.json a file name null.d is created

adiazulay, 2021-02-20:
I ran into this testing on MacOS. @iFreilicht any idea what might be causing this?

iFreilicht, 2021-02-22:
No, but I can reproduce this on Linux. What contents does the file have on your machine? On mine it reads:

    null.o: /dev/null

iFreilicht, 2021-02-22:
So it seems this file is generated only when the project is loaded.
If you delete it, then it won't reappear until you reload the window, no matter what you do. Even then, that's not 100% reproducible.
The only sure way to get this bug is to close the workspace and open the project again.

adiazulay, 2021-02-22:
```null.d``` is the same on MacOS

The file is created even with the Arduino extension disabled, so I think this is actually a bug with IntelliSense

iFreilicht, 2021-02-28:
Huh, that is super weird. It could also be related to the C/C++ extension and the long `c_cpp_properties.json`, as I've never seen this happen in any Rust or Python workspace.

adiazulay, 2021-03-01:
So I narrowed it down to specifically the ```-MMD``` complier arg that's causing the issue.

Is it strictly necessary to include all complier args in ```c_cpp_properties.json```?

iFreilicht, 2021-03-01:
Huh, I stumbled across that as well, should've dug deeper. We could leave out `-MMD` specifically. The idea of the extension was to mirror the environment of the Arduino IDE as closely as possible, but it seems people have always built fine without it. The only other ticket even mentioning this flag is #765 and even there it only appeared as something copied over from elsewhere.

Let's try it without `-MMD`, and if the examples build properly, we'll leave it out for good.

adiazulay, 2021-03-01:
I just checked the documentation on ```-MMD``` creates make file dependencies, so it makes sense that it's creating a file a ```.d``` file. It looks like during the library detection part of the build Arduino does pass this argument ```-o /dev/null```

I think it's fine to remove it since it seems that everything is working fine without it, we should document this in case it needs to be added back in at some point.

adiazulay, 2021-03-04:
@iFreilicht #1208 should fix this. It's a bit brute force, but it works.

PR #1208 (2021-03-05): remove -MMD from complier args for intellisesne
```-MMD``` complier arg creates a file named null.d in the workspace directory. This PR removed the argument from intelliSense configuration.

Resolves #1206
closed 2021-03-10 by adiazulay