microsoft/vscode-arduino #1215: Output monitor don't shows sketch memory usage

lozziboy, 2021-03-24:
[image]

As described in the title, after compiling the sketch output monitor don't shows sketch memory usage like in the IDE environment

cth000182, 2021-03-24:
You can change your Log Level in the settings under the Arduino Extension. It should be "Info" but you can change to "Verbose" - you have to search a bit, but the usage is there.

lozziboy, 2021-03-25:
> You can change your Log Level in the settings under the Arduino Extension. It should be "Info" but you can change to "Verbose" - you have to search a bit, but the usage is there.

That's true.
However it's a big disadvantage the fact that this important information is drowned among hundreds of others, so there is the needs to spend time in searching for it

cth000182, 2021-03-25:
@lozziboy I completely agree! We ran into this yesterday and were wondering why our compiler was not showing the normal information during a build - thought we broke something! Did not know it was an update... 

Hopefully, they can get this resolved as showing the usage is quite important when writing code. If we find a method, I will post here.

adiazulay, 2021-03-26:
Thanks for catching this bug, this likely has to do with how the IntelliSense generates the settings. Will look into this for the next release.

iqapps, 2021-05-31:
Reminder: We got a new release since the last comment here

dasfdlinux, 2021-06-16:
Still missing this percent usage.

dealmeal, 2021-08-28:
> Thanks for catching this bug, this likely has to do with how the IntelliSense generates the settings. Will look into this for the next release.

Please 🙏

dasfdlinux, 2021-10-05:
@adiazulay Can this be marked as a regression and moved higher up in the priority list? At the moment I'm writing my source in Code, and then going to my other monitor where I have the Arduino IDE pulled up for compiling so I can get the program space utilization.

Not to be a grouch, but this has been marked as an issue for over six months. And it's not a feature request, it was the result of a breaking change.

jelimoore, 2021-11-06:
+1. This is a necessary feature for me and not having it is making me slowly crawl back to the Arduino IDE.

dasfdlinux, 2022-01-13:
Thank you, @AlexIII and @benmcmorran!

benmcmorran, 2022-01-15:
This issue has been fixed in the latest release of this extension, which is available in the VS Code extension marketplace.

closed 2022-01-15 by benmcmorran

SirSydom, 2022-02-18:
memory usage is not displayed with 0.4.11

lozziboy, 2022-02-18:
> memory usage is not displayed with 0.4.11

That's not true in my case
[image]

BelKed, 2022-02-18:
Probably it's related to this issue:
- https://github.com/microsoft/vscode-arduino/issues/1417

hicwic, 2022-04-27:
thx, English language on arduino IDE does the trick. Was in french, changed to English, restarted VSCode, et memory report is displaying. Thx you

hydrafi, 2023-02-18:
Yes it works, but only if IDE used. Now Ide is deprecated .... with cli the report is displayed only if verbose, but there are too much info... with info flag the report isn't displayed ... is there a solution?

SimonArrowhead, 2024-02-05:
Is there any progress in resolving this issue?

matteolavaggi, 2024-03-23:
Problem still here with last version of VSC + Arduino extension using Arduino CLI and Arduino IDE in english:

`Writing | ################################################## | 100% 0.14s

avrdude: 848 bytes of flash written

avrdude done.  Thank you.

IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Cmd+Alt+I"
[Done] Uploading sketch 'ArduinoCopilot.ino'`

commit 0295850b (2022-01-15): Fix #1215 Output sketch memory usage in non-verbose mode
commit c5b03c9c (2022-01-15): Merge pull request #1383 from AlexIII/fix_no_verbose_sketch_stats_output

Fix #1215 Output sketch memory usage in non-verbose mode