microsoft/vscode-arduino #1216: identifier "Serial" is undefined with version 0.4.0

lozziboy, 2021-03-25:
This is to open again the same issue discussed here
https://github.com/microsoft/vscode-arduino/issues/866
and there
https://github.com/microsoft/vscode-arduino/issues/808

IntelliSense has still some problem in finding the following keywords 
"Serial"
"TCCR2A"
"TCCR2B"
"OCIE0A"

To manually solve this, I added the following defines to c_cpp_properties.json after disabling Intelli Sense Auto Gen from the vscode extension settings
"USBCON",
  "__AVR_ATmega328P__"

adiazulay, 2021-03-26:
This is odd, I tested this with the uno and it was working before we released. 

I'm able to repro this on my Machine now, looking into what changed.

zeroamps, 2021-04-03:
Hi guys, 
I can confirm the same issue on a WIN10 machine.
[image]

adiazulay, 2021-04-05:
so it looks like removing the "-MMD" complier arg from c_cpp_properties.json is culprit. We removed it because it was creating file erroneously on mac and linux.

Will revert to including it in the next release

iFreilicht, 2021-04-07:
And we were like "What could possibly go wrong?" Good find! Maybe we can find another solution for the `null.d`-problem then.

adiazulay, 2021-04-13:
#1231 forces USBCON to be included in all config files.

adiazulay, 2021-04-15:
This should be fixed in v0.4.1 do you mind giving it a try and letting me know if it works?
https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.1-rc1

ccramer, 2021-04-15:
Awesome, workes like a charm for me now with v0.4.1. It added the "USBCON".

coertg, 2021-04-17:
Hoping this update will ship soon!

Peter707-try, 2021-04-18:
0.4.1.rc1 works fine for me: USBCON is automatically added to c_cpp_properties.json
The Serial (not defined) issue is solved.
Thank You very much :) +1:

lozziboy, 2021-04-19:
> This should be fixed in v0.4.1 do you mind giving it a try and letting me know if it works?
> https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.1-rc1

Unfortunately the problem is still partially present.

Updated to 0.4.1-rc1
[image]

Still having problems with other keywords
[image]

To workaround this, I added the following define to c_cpp_properties.json:
"__AVR_ATmega328P__"

iFreilicht, 2021-04-19:
Thank you so much for the screenshot! Now I realize why we didn't see this during testing, we only ever tried using the Arduino API, not the chip registers. Can you check whether the `-MMD` flag is added to the compiler options by the auto generation?

lozziboy, 2021-04-19:
> Thank you so much for the screenshot! Now I realize why we didn't see this during testing, we only ever tried using the Arduino API, not the chip registers. Can you check whether the `-MMD` flag is added to the compiler options by the auto generation?

Where can I check it?

adiazulay, 2021-04-19:
@iFreilicht I looked into it, and instead of adding back `-MMD` I added `USBCON `to defines. Which reliably fixes the serial issues.
@lozziboy I'm going to take second part of this as a backlog item and open a new issue for it.

yszty, 2021-05-10:
Hello, same problem with EEPROM

[image]
[image]
[image]

`{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Program Files (x86)\\Arduino\\tools\\**",
                "C:\\Users\\%userName%\\Documents\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\**",
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "defines": [
                "USBCON"
            ],
            
            "intelliSenseMode": "windows-msvc-x64",
            "cStandard": "c17",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}`

lozziboy, 2021-06-01:
> Hello, same problem with EEPROM
> 
> [image]
> [image]
> [image]
> 
> ` "defines": [ "USBCON" ],`

Hello,
I don't have this problem with EEPROM.

Did you installed EEPROM library from IDE?
Please check if you have EEPROM library include path in file  c_cpp_properties.json
[image]

itopaloglu83, 2021-12-02:
Okay. Here's another strange behavior. Defining `"USBCON", "__AVR_ATmega328P__"`  causes the following error on Arduino.h line 326.

> #error directive: "Targets with both UART0 and CDC serial not supported"C/C++(35)

Maybe not every board by default require USBCON. I added the following code block to my sketch to check if USBCON is defined at compile time at it wasn't.

```
#ifdef USBCON
#error "What, what?"
#endif
```

iFreilicht, 2021-12-02:
That is by design. As the Atmega328 doesn't have any USB interface, you can't define USBCON. The USB connectivity on an Arduino Uno is provided by a USB-to-Serial converter, and to the Atmega328 it just looks like a UART connection.
USBCON is automatically defined on all boards that have an MCU with a built-in USB interface, like the Arduino micro. Here, the CDC connection is directly exposed to the MCU. You can never have both UART and CDC as the primary serial interface.

itopaloglu83, 2021-12-02:
@iFreilicht 

The v0.4.1 release is manually adding the `USBCON` to all boards, regardless of their capabilities. That is the issue I'm trying to show.

https://github.com/microsoft/vscode-arduino/blob/7ef8b8f4bba788aa3f8a22c322ca99c03e752a4e/src/arduino/intellisense.ts#L105

itopaloglu83, 2021-12-02:
For anyone using Arduino Uno. After the `c_cpp_properties.json` file is created, just disable to automated update and add the following definitions to it. These are coming from `AVR-GCC` command line parameter of `-mmcu=atmega328p`. The details can be found on https://www.nongnu.org/avr-libc/user-manual/using_tools.html website where options are explained.

```
// AVR-GCC is used.
"AVR=1",
"__AVR=1",

// The architecture of ATmega328P is AVR5.
"__AVR_ARCH__=5",
"__AVR_MEGA__",
"__AVR_ENHANCED__",
"__AVR_HAVE_JMP_CALL__",
"__AVR_HAVE_MOVW__",
"__AVR_HAVE_LPMX__",
"__AVR_HAVE_MUL__",
"__AVR_2_BYTE_PC__",

// The MCU type is ATmega328P.
"__AVR_ATmega328P__",

// Additional command line define statements.
"F_CPU=16000000L",
"ARDUINO=10607",
"ARDUINO_AVR_UNO",
"ARDUINO_ARCH_AVR"

```

iFreilicht, 2021-12-03:
Oh boy. Looking at the history of this ticket, forcing USBCON was supposed to work around IntelliSense not finding Serial, but if USBCON can't be used on some devices, then that workaround only works inside IntelliSense.

I think we have to go back to using `-MMD` and fix the issue of null.d getting created on Unix instead. That is likely more reliable than stacking workarounds on top of workarounds.

itopaloglu83, 2021-12-11:
I later realized that adding `"-mmcu=atmega328p"` to the `compilerArgs` part also resolves the errors for Arduino Uno as well. 

```json
{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "/home/pi/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc",
            "compilerArgs": [
                "-w",
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing",
                // Microcontroller: ATmega328P
                "-mmcu=atmega328p"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "/home/pi/.arduino15/packages/arduino/hardware/avr/1.8.4/cores/arduino",
                "/home/pi/.arduino15/packages/arduino/hardware/avr/1.8.4/variants/standard",
                "/home/pi/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/lib/gcc/avr/7.3.0/include",
                "/home/pi/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/lib/gcc/avr/7.3.0/include-fixed",
                "/home/pi/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include"
            ],
            "forcedInclude": [
                "/home/pi/.arduino15/packages/arduino/hardware/avr/1.8.4/cores/arduino/Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                // Arduino Uno Specific Definitions
                "F_CPU=16000000L",
                "ARDUINO=10607",
                "ARDUINO_AVR_UNO",
                "ARDUINO_ARCH_AVR"
            ]
        }
    ]
}
```

apws, 2022-01-17:
hi guys, I just stared to test vscode+arduino and have problem with intellisense around Serial too, but only using "megaTinyCore" boards-package and Microchip Curiosity Nano Tiny1627 - on original Leonardo and Infineon XMC2Go (cortex m0+) board packages, it doesnt happen. Code normally builds/uploads/runs, but there is only reported problem... that board uses nEDBG OB programmer (XMC2Go uses segger J-link lite).

rednebmas, 2022-01-29:
Add the following **above** your `#include <Arduino.h>` to get the error to go away. 

```cpp 
#define USBCON 1 // Prevents Serial errors from VSCode
```

The reason is that Serial _is defined_, in CDC.cpp (in arduino/avr/cores/arduino) but the definition for Serial is hidden behind an `#if defined(USBCON)`.

lozziboy, 2022-02-23:
> ```c++
> #define USBCON 1
> ```

This don't works with the chip registers
[image]