microsoft/vscode-arduino #1221: Plugin appends 'Arduino.app' to path on MacOS

recursivecodes, 2021-03-31:
Why? 

If I'm using the CLI, I put `/Users/bin/` and `arduino-cli` for the command path, but it ends up looking in `/Users/bin/Arduino.app/arduino-cli`. If I'm trying to use the CLI, it should not append anything to the path like that.

https://github.com/microsoft/vscode-arduino/blob/3bb7b14f1dcf803180e6a16773dccebc9ced6725/src/common/sys/darwin.ts#L13

recursivecodes, 2021-04-09:
Any thoughts on this?

awmichel, 2021-04-11:
@recursivecodes I ran into the same issue and I believe the PR I just opened #1227 resolves this issue. I'm not too familiar with VSCode extension development, so I'm not sure about the steps to install an extension from a git repo, but I manually edited the source in `~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.0/out/src/` to test out my changes.

adiazulay, 2021-04-13:
Thanks @awmichel for fixing this!

@recursivecodes this was an artifact of how the extension would handle accessing the IDE in older builds.

adiazulay, 2021-04-15:
This should be fixed in v0.4.1. @awmichel do you mind giving this a testing it out and letting me know if it works?
https://github.com/microsoft/vscode-arduino/releases/tag/v0.4.1-rc1

awmichel, 2021-04-16:
@adiazulay Installed 0.4.1-rc1 locally and was able to verify and upload to a ESP32 using `arduino-cli`.

For anyone else using a MacOS M1 w/ `arduino-cli` installed via Homebrew, your VSCode settings should look something like:
```json
{
    "arduino.useArduinoCli": true,
    "arduino.path": "/opt/homebrew/bin/",
    "arduino.commandPath": "arduino-cli",
}
```

On a non-M1 Mac (untested):
```json
{
    "arduino.useArduinoCli": true,
    "arduino.path": "/usr/local/bin/",
    "arduino.commandPath": "arduino-cli",
}
```

PR #1227 (2021-04-12): Fix arduinoPath resolution on MacOS
Fixes #1221 

The `resolveMacArduinoAppPath` utility was automatically adding "Arduino.app" to the resolved `arduinoPath` regardless of whether `useAndroidCli` was enabled.

This was breaking installations of `arduino-cli` made through Homebrew. The path to the command ends up being either `/usr/local/bin/arduino-cli` or `/opt/homebrew/bin/arduino-cli`. This function would resolve an `arduinoPath` of `/opt/homebrew/bin` to `/opt/homebrew/bin/Arduino.app/arduino-cli`.

The contributing guidelines mention opening PRs against the develop branch, but that branch [... cut]
closed 2021-04-20 by adiazulay