microsoft/vscode-arduino #1223: Version 4.0 does not work with the board selection.

vde69, 2021-04-03:
Version 4.0 does not work with the board selection.

Windows-10,
Arduino-1.8.13

[image]

Version 3.5 works.

-------------------------------
c_cpp_properties.json :

           "includePath": [
                "${workspaceFolder}/**",
                "C:\\Program Files (x86)\\Arduino\\**",
                "C:\\Users\\D1\\AppData\\Local\\Arduino15\\**",
                "C:\\Users\\D1\\Documents\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\libraries\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\**",
                "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\**",
                "C:\\Program Files (x86)\\Arduino\\tools\\**"
            ],
            "forcedInclude": [
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\Arduino.h"
            ],
            "defines": [
                "UNICODE",
                "_UNICODE",
                "ARDUINO=160",
                "USBCON"
            ],
            "compilerPath": "C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr\\bin\\avr-gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "intelliSenseMode": "gcc-x64"

-------------------------------

adiazulay, 2021-04-12:
is the Arduino CLI option selected by any chance?

vde69, 2021-04-14:
If you disable the "Arduino CLI" option, it works. 
But it is not clear why it does not work with the enabled option, because the Arduini IDE is installed as standard path.

adiazulay, 2021-04-14:
If you select Arduino CLI option it will try to use CLI rather than the IDE. So even if you have the IDE installed it will fail to find the CLI executable in the folder.