microsoft/vscode-arduino #1224: Current Open Sketch not Selected

momiccioli, 2021-04-04:
Hello,

this tool almost works for me and I would love to continue to use.  The only thing preventing me from using it is that I have to manually select the sketch that I want verify/uploaded.  I prefer that when I select the Verify or Upload it does this based on the current sketch that I have open. I work on many sketches at the same time so I switch often and this feature would be helpful.

Currently I use a make file and buttons that run scripts to accomplish what I just explained.  While this works well your solution has advantages over mine except the one issue I have highlighted.  If by chance that functionality is there please point me in the right direction and I will gladly try the product further.

Otherwise nice solution and keep at it.

matthboers, 2022-03-26:
This has been a bit of a pain for me as well. Wish there was a option in the extensions settings to have the `Verify` and `Upload` buttons use the currently opened sketch, and not the last manually selected sketch. This would make the behaviour more congruent compared to the Arduino IDE.

In the meantime, I have found this older Issue (#807), [where Iacerto linked a script](https://github.com/microsoft/vscode-arduino/issues/807#issuecomment-571067589) which would clear the currently selected sketch from the `arduino.json` file. From what I gather, this makes it so that you have to manually reselect the sketch you want to compile every time, but it's better than debugging a sketch you have not even uploaded! 😉

BelKed, 2022-03-26:
If am I correct, there’s already opened a pull request that will add option for uploading and verifying sketch in active tab:
- https://github.com/microsoft/vscode-arduino/pull/964

However, I have no idea why it’s not yet merged…

ChitoKim, 2023-11-01:
I found a vscode extension exactly for this. Basically it's kinda add-on, because arduino extension is its dependency.
Also you can install it on vscode extension search : "Arduino Sketch Auto Switcher" 
Finally I can stop debugging a code I was looking for.
https://github.com/nosknut/arduino-sketch-auto-switcher