microsoft/vscode-arduino #1228: cannot open source file "avr32/io.h"

fletchjp, 2021-04-12:
I am working with VS Code and the Arduino extension. When I attempt to use a file which include <SPI.h> it fails with an error saying cannot open source file "avr32/io.h". The file works in the Arduino IDE. Several people have reported a similar error on the web and suggest that it is a problem with the ordering of include files.

adiazulay, 2021-04-12:
What version of the extension are you using?

vde69, 2021-04-12:
This problem occurs when libraries have files with names like in the IDE, that is, when linking, it picks up the first file found by name, without taking into account the priorities of the local location of the source. The correct solution would be to first search in the directory hierarchy of the current file and then in the list of included directories. I have this problem occurred when connecting ESP 01, I decided to rearrange the directories.

fletchjp, 2021-04-12:
I am using Arduino 0.4.0 and C/C++ 1.3.0-insiders5

fletchjp, 2021-04-12:
I have the error with the example DigitalPotControl which comes with the SPI library to be found in Arduino->hardware->arduino->avr->libraries. I can find nothing to help with this in the Arduino reference for SPI at https://www.arduino.cc/en/Reference/SPI. 

The folder Arduino->hardware->tools->avr->avr->include->avr  has io.h

I cannot find avr32 anywhere.

fletchjp, 2021-04-13:
I think I have resolved this using the work documented here:
https://github.com/Microsoft/vscode-arduino/issues/678
The issue seems to be getting the include files correct.
The critical lines seem to be these, in the Win32 configuration:

                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\**", 
                "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src", 

The problem is that Intellisense does not generate them. In my main problem case SPI.h was not a top level include.

fletchjp, 2021-04-13:
It also seems to be necessary for this to be searched first:

                "C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\**",

ed-graham, 2022-04-21:
I think I have worked out what is going on here.  I've written a full explanation under [issue 678](https://github.com/Microsoft/vscode-arduino/issues/678#issuecomment-1104975628).