microsoft/vscode-arduino #1230: Upgrading a Board Breaks The Project

btribley, 2021-04-13:
I am using @SpenceKonde DxCores. Fantastic core. It was at version 1.3.2, tonight my Arduino IDE (used for compiling because VS Code setup simply doesn't work out of the box...) let me know that version 1.3.4 was available so I installed it.

Instantly my VS Code Arduino project became totally broken. It can't find Arduino.h and avr/*.h .

I recognized that most likely the board update in the Arduino IDE didn't translate. So, I went back and re-selected the board, no joy. It seems that the Arduino extension will not load an upgrade. 

c_cpp_properties.json still shows all the version 1.3.2 stuff even though I have restarted both the Arduino IDE and VS Code several times. 

I cannot see any way to reload my board to pick up the update. So I will do the hokey thing of manually patching the json to upgrade. 

Am I missing something?

SpenceKonde, 2021-04-13:
v1.3.4 is contains a critical defect and I'm in the process of rekitting
the windows toolchain package for a 1.3.5 - the directory structure of the
toolchain packages is a work of inspired evil; the IDE is also not smart
enough to recognize when there's an extra level of directories (this made a
mess in ATTinyCore 1.5.0 as well), but the effect of it should only be to
prevent compiling from working on Windows platforms (avr-gcc not found),
and other users reported success using the IDE to move back to 1.3.2
(somehow this had gone unnoticed and unreported until today despite being
in the 1.3.3 and 1.3.4 packages that came out last week - 1.3.4 caught a
similar issue on Linux/Mac OS, but I thought windows was OK. It was only
discovered when I included the same bad component in in megaTinyCore 2.3.0
- and then someone noticed, because there the newly added features are
support for new parts with an entirely new ADC, and the ability to both
make use of that and take full advantage of the existing ADC on the older
parts, so people are a lot more excited for the next release of that, as
opposed to DxCore, where the release was largely "How did so many changes
build up? I haven't done anything!" and I was getting worried about the
following release bringing it too much at once.  DxCore will gets
it's version of analogReadEnh() and analogReadDiff() in the next release
after 1.3.5 (that is - before the DD-series support planned for 1.4.0
pending Microchip shipping the damned things), though the differential ADC
hardware on the Dx-series leaves something to be desired. The tinyAVR
2-series is the "trial balloon" for the next generation ADC, which goes
into the EA-series full-size parts.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

btribley, 2021-04-13:
While the core has defects VS Code blithely ignored the upgrade. For example, the json file still said:
               "DXCORE=\"1.3.2\"",
after the upgrade to 1.3.4 . I am pretty sure that 1.3.5 will still not update in VS Code, but I will report back after installing it.  I bet Arduino doesn't update their board cores much, particularly for the mature parts. So the VS Code people haven't had much of a reason to test an upgraded Arduino core.

SpenceKonde, 2021-04-13:
mmm depends how long vscode has been doing this for. There was a burst in
the 1.6.x versions with a lot of rapid fire releases of defective builds.
Not quite like my cores but way faster than official core has been doing
recently. The AVR core version gets bumped every IDE release whether it
needs it or not and I think most of those were more about ways in which the
rest of the IDE was broken, though the core wasn't innocent either.
Now that I think about it, the official megaavr core is way overdue for a
release, just judging by how many bugs I know were fixed in github since
the last release...

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

adiazulay, 2021-04-13:
@btribley Are you using v0.4 of the extension?

If you are can you try to rebuild intellisense configuration (CTRL  + ALT + I)

btribley, 2021-04-14:
Hi Adi,

It turns out that it works if the core package is internally consistent. Intellisense rebuilds when VS Code opens. Thanks for pointing this out!

Bill

---- On Tue, 13 Apr 2021 12:08:15 -0400 Adi Azulay ***@***.***> wrote ----

https://github.com/btribley Are you using v0.4 of the extension?

If you are can you try to rebuild intellisense configuration (CTRL  + ALT + I)

closed 2021-04-14 by btribley

SpenceKonde, 2021-04-14:
So was the problem just because 1.3.4 was busted?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

btribley, 2021-04-14:
Yes. See my comment on the closed issue regarding 1.3.3 and 4. There may still be a problem although the tools work now. Debugging my own stuff that has nothing to do with the core!