microsoft/vscode-arduino #1232: Cannot select serial port, open serial port (VSCode, Arduino, Apple M1)

khorbushko, 2021-04-15:
Looks like the issue is can be reproduced on new M1 from Apple - can't select neither port neither open serial port (if manually write it in config). Upload works fine. 

When clicked on "open port"button
[image]

_Originally posted by @khorbushko in https://github.com/microsoft/vscode-arduino/issues/1099#issuecomment-819746181_

philffm, 2021-04-25:
Second this. Tried to manually adjust the port in arduino.json file but it fails at opening the port 🤷🏽‍♂️

khorbushko, 2021-04-25:
for me, manual configuration in of `arduino.json`works fine:

```
{
    "sketch": "serial.ino",
    "board": "arduino:avr:uno",
    "programmer": "arduino:usbasp",
    "port": "/dev/cu.usbmodem101"
}
```

To get serial port - u can use Arduino IDE or IOKit and Xcode

philffm, 2021-04-25:
Thanks for letting me know.

It still throws the following error

`Failed to open serial port /dev/cu.usbserial-11430 due to error: + Error: dlopen(/Users/user/.vscode-insiders/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/usb-detection/build/Release/detection.node, 1): no suitable image found. Did find: /Users/user/.vscode-insiders/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/usb-detection/build/Release/detection.node: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00 /Users/user/.vscode-insiders/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/usb-detection/build/Release/detection.node: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00`

khorbushko, 2021-04-25:
yes, this is reproducible on my side as well, setting json config is just kind of workaround... neither port selection, neither connection to serial port works on my side. In same moment, in Arduino IDE everything works fine. I also created a small utility for Serial port scan and communication using IOKit - and it works just fine too. So it's looks like completely issue related to this extension.

Yatheesha1, 2021-04-27:
for me also, manual configuration in of arduino.json works fine, but I am unable to open the serial.monitor in vscode

robbiewu008, 2021-05-06:
The usb detection module's native binary is not for aarch64, is there one detection.node for aarch64, or can we compile one?

robbiewu008, 2021-05-06:
After checked the upstream projects usb detection, still no official release for aarch64, I create one issue # https://github.com/MadLittleMods/node-usb-detection/issues/141

robbiewu008, 2021-05-06:
brew's node.js@16, NODE_MODULE_VERSION=93; brew's node.js@14, NODE_MODULE_VERSION=83; but the node.js embedded in vscode is NODE_MODULE_VERSION=87, so encounter new issues:
---------------
Failed to open serial port /dev/cu.usbserial-0001 due to error: + Error: The module '/Users/robbiewu/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/usb-detection/build/Release/detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 93. This version of Node.js requires
NODE_MODULE_VERSION 87. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

Failed to open serial port /dev/cu.usbserial-0001 due to error: + Error: The module '/Users/robbiewu/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.2/out/node_modules/usb-detection/build/Release/detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 83. This version of Node.js requires
NODE_MODULE_VERSION 87. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

robbiewu008, 2021-05-06:
After more debugging, the up-up stream project node-gyp's issue, I created one issue #https://github.com/nodejs/node-gyp/issues/2391.

closed 2023-04-02 by github-actions[bot]