microsoft/vscode-arduino #1234: Unable to upload using programmer USBTinyISP - MacOS

felipesolish, 2021-04-18:
When trying to upload using programmer I'm getting this error

Error: Can only pass one of: --upload, --verify, --get-pref, --install-boards, --version, --install-library
[Warning] Failed to generate IntelliSense configuration.
[Error] Uploading (programmer) sketch 'PotSpeedControl.ino': Exit with code=3

-----
Full command log is:

```
2021-04-18 12:42:14.864 Arduino[68116:4021836] Loading Application 'Arduino'
2021-04-18 12:42:14.865 Arduino[68116:4021836] JVMRuntime=jre8u252-b09.jre
2021-04-18 12:42:14.865 Arduino[68116:4021836] CFBundleName=Arduino
2021-04-18 12:42:14.865 Arduino[68116:4021836] WorkingDirectory=(null)
2021-04-18 12:42:14.865 Arduino[68116:4021836] JVMMainClassName=processing.app.Base
2021-04-18 12:42:14.865 Arduino[68116:4021836] JVMOptions=(
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
2021-04-18 12:42:14.865 Arduino[68116:4021836] JVMArguments=(
)
2021-04-18 12:42:14.865 Arduino[68116:4021836] JVMClasspath=(null)
2021-04-18 12:42:14.865 Arduino[68116:4021836] JVMDefaultOptions={
}
2021-04-18 12:42:14.865 Arduino[68116:4021836] -> Bundle path: /Applications/Arduino.app
2021-04-18 12:42:14.865 Arduino[68116:4021836] -> Working Directory: '/Users/felipesolis/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.0/out/src'
2021-04-18 12:42:14.865 Arduino[68116:4021836] -> JVM Runtime path: /Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre
2021-04-18 12:42:14.865 Arduino[68116:4021836] Searching for a Java 8 virtual machine
2021-04-18 12:42:14.865 Arduino[68116:4021836] Search for java VM in '/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home'
2021-04-18 12:42:14.866 Arduino[68116:4021836]   KO - error: 'launch path not accessible'
2021-04-18 12:42:14.935 Arduino[68116:4021836] Search for java VM in '/Library/Java/JavaVirtualMachines/jdk-16.jdk/Contents/Home'
2021-04-18 12:42:15.195 Arduino[68116:4021836]   KO - invalid 'java -version' output
2021-04-18 12:42:15.195 Arduino[68116:4021836] -> Java Runtime Dylib Path: '/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/lib/jli/libjli.dylib'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Command line passed to application argc=34:
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 0: '/Applications/Arduino.app/Contents/MacOS/Arduino'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 1: '-Djava.class.path=/Applications/Arduino.app/Contents/Java/log4j-core-2.12.0.jar:/Applications/Arduino.app/Contents/Java/rsyntaxtextarea-3.0.3-SNAPSHOT.jar:/Applications/Arduino.app/Contents/Java/batik-ext-1.8.jar:/Applications/Arduino.app/Contents/Java/slf4j-api-1.7.22.jar:/Applications/Arduino.app/Contents/Java/arduino-core.jar:/Applications/Arduino.app/Contents/Java/batik-xml-1.8.jar:/Applications/Arduino.app/Contents/Java/slf4j-simple-1.7.22.jar:/Applications/Arduino.app/Contents/Java/batik-dom-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-compress-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-httpclient-3.1.jar:/Applications/Arduino.app/Contents/Java/bcprov-jdk15on-152.jar:/Applications/Arduino.app/Contents/Java/commons-logging-1.0.4.jar:/Applications/Arduino.app/Contents/Java/log4j-api-2.12.0.jar:/Applications/Arduino.app/Contents/Java/batik-script-1.8.jar:/Applications/Arduino.app/Contents/Java/jackson-annotations-2.9.5.jar:/Applications/Arduino.app/Contents/Java/batik-parser-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-squiggle-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-awt-util-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-gvt-1.8.jar:/Applications/Arduino.app/Contents/Java/bcpg-jdk15on-152.jar:/Applications/Arduino.app/Contents/Java/xml-apis-1.3.04.jar:/Applications/Arduino.app/Contents/Java/batik-anim-1.8.jar:/Applications/Arduino.app/Contents/Java/jtouchbar-1.0.0.jar:/Applications/Arduino.app/Contents/Java/batik-bridge-1.8.jar:/Applications/Arduino.app/Contents/Java/jssc-2.8.0-arduino4.jar:/Applications/Arduino.app/Contents/Java/batik-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-transcoder-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-svg-dom-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-rasterizer-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-codec-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-io-2.6.jar:/Applications/Arduino.app/Contents/Java/commons-codec-1.7.jar:/Applications/Arduino.app/Contents/Java/xmlgraphics-commons-2.0.jar:/Applications/Arduino.app/Contents/Java/apple.jar:/Applications/Arduino.app/Contents/Java/commons-exec-1.1.jar:/Applications/Arduino.app/Contents/Java/batik-css-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-util-1.8.jar:/Applications/Arduino.app/Contents/Java/jna-platform-4.2.2.jar:/Applications/Arduino.app/Contents/Java/commons-net-3.3.jar:/Applications/Arduino.app/Contents/Java/batik-svgpp-1.8.jar:/Applications/Arduino.app/Contents/Java/xml-apis-ext-1.3.04.jar:/Applications/Arduino.app/Contents/Java/jmdns-3.5.5.jar:/Applications/Arduino.app/Contents/Java/java-semver-0.8.0.jar:/Applications/Arduino.app/Contents/Java/pde.jar:/Applications/Arduino.app/Contents/Java/jna-4.2.2.jar:/Applications/Arduino.app/Contents/Java/commons-lang3-3.8.1.jar:/Applications/Arduino.app/Contents/Java/jackson-core-2.9.5.jar:/Applications/Arduino.app/Contents/Java/jsch-0.1.50.jar:/Applications/Arduino.app/Contents/Java/jackson-databind-2.9.5.jar'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 2: '-Djava.library.path=/Applications/Arduino.app/Contents/MacOS'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 3: '-DLibraryDirectory=/Users/felipesolis/Library'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 4: '-DDocumentsDirectory=/Users/felipesolis/Documents'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 5: '-DApplicationSupportDirectory=/Users/felipesolis/Library/Application Support'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 6: '-DCachesDirectory=/Users/felipesolis/Library/Caches'
2021-04-18 12:42:15.203 Arduino[68116:4021836] Arg 7: '-DSandboxEnabled=true'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 8: '-Dapple.awt.application.name=Arduino'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 9: '-Dcom.apple.macos.use-file-dialog-packages=true'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 10: '-Dcom.apple.smallTabs=true'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 11: '-DAPP_DIR=/Applications/Arduino.app/Contents/Java'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 12: '-Djava.ext.dirs=/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/lib/ext/:/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/jre/lib/ext/'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 13: '-Djava.net.preferIPv4Stack=true'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 14: '-Xdock:name=Arduino'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 15: '-Dcom.apple.mrj.application.apple.menu.about.name=Arduino'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 16: '-Dfile.encoding=UTF-8'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 17: '-Xms128M'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 18: '-Xmx512M'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 19: '-splash:/Applications/Arduino.app/Contents/Java/lib/splash.png'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 20: 'processing.app.Base'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 21: '--board'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 22: 'arduino:avr:uno'
2021-04-18 12:42:15.204 Arduino[68116:4021836] Arg 23: '--upload'
2021-04-18 12:42:15.205 Arduino[68116:4021836] Arg 24: '--useprogrammer'
2021-04-18 12:42:15.205 Arduino[68116:4021836] Arg 25: '--pref'
[... 55 lines cut ...]
2021-04-18 12:42:15.265 Arduino[68116:4021878] Arg 13: '-Djava.net.preferIPv4Stack=true'
2021-04-18 12:42:15.265 Arduino[68116:4021878] Arg 14: '-Xdock:name=Arduino'
2021-04-18 12:42:15.265 Arduino[68116:4021878] Arg 15: '-Dcom.apple.mrj.application.apple.menu.about.name=Arduino'
[... cut ...]
2021-04-18 12:42:15.266 Arduino[68116:4021878] Arg 32: 'build.path=/Users/felipesolis/Desktop/SophiesMusicPlayer/ArduinoOutput'
2021-04-18 12:42:15.266 Arduino[68116:4021878] Arg 33: '/Users/felipesolis/Desktop/SophiesMusicPlayer/PotSpeedControl/PotSpeedControl.ino'
Error: Can only pass one of: --upload, --verify, --get-pref, --install-boards, --version, --install-library
[Warning] Failed to generate IntelliSense configuration.
[Error] Uploading (programmer) sketch 'PotSpeedControl.ino': Exit with code=3
'''

drewlsvern, 2021-05-07:
@felipesolish Did you ever find a resolution?

drewlsvern, 2021-05-07:
@felipesolish I managed to find a workaround.

I had to add the following to my settings.json in vscode:
`
    "arduino.path": "C:\\Program Files (x86)\\Arduino",
    "arduino.commandPath": "arduino-cli.exe",
    "arduino.useArduinoCli": true
`

You will need to download the arduino cli [here](https://github.com/arduino/arduino-cli/releases/tag/0.13.0)

I only linked to the older version 0.13.0 as the README states it's the only tested version with this extension.

jeffrson, 2021-05-21:
Have the same problem with AVRISP mkII.

BTW:
Trying to select programmer just shows an empty input line. No log found in build path (other than stated in message).

elektrobil, 2021-07-03:
I have the same problem.

rrakso, 2021-10-05:
Me to!

AqeeAqee, 2022-01-05:
Where can I found ~/.vscode/extensions/vsciot-vscode.vscode-arduino-0.4.7/out/src/arduino/arduino.js
in windows?
Thanks 
@rrakso @a7hybnj2

benmcmorran, 2022-01-13:
A fix for this issue from #1369 will be included in the next extension release.

rrakso, 2022-01-14:
Hi @AqeeAqee, sorry for the late reply.

> Extensions are installed in a per user extensions folder. In Windows, the location is in the following folder `%USERPROFILE%\.vscode\extensions`

_Source: [Common questions > Where are extensions installed?](https://code.visualstudio.com/docs/editor/extension-marketplace#:~:text=Extensions%20are%20installed%20in%20a,vscode%5Cextensions)_

Happy Coding :grin:

benmcmorran, 2022-01-15:
This issue has been fixed in the latest release of this extension, which is available in the VS Code extension marketplace.

closed 2022-01-15 by benmcmorran