microsoft/vscode-arduino #1235: [Warning] Failed to generate IntelliSense configuration.

deepcoder, 2021-04-18:
What do I need to do to correct this on MacOS 10.4.6 with : 
```
Version: 1.56.0-insider
Commit: ce791d27dc9889a392f4d29abefe9c53eb0b5129
Date: 2021-04-16T11:23:31.851Z
Electron: 11.4.2
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 18.7.0

and 
Arduino 1.8.13

```

calebho, 2021-04-20:
Having the same issue
```
Version: 1.55.2
Commit: 3c4e3df9e89829dce27b7b5c24508306b151f30d
Date: 2021-04-13T09:36:32.643Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 20.3.0
```
Extension configuration:
```
    "arduino.path": "/usr/local/bin/arduino-cli",
    "arduino.defaultBaudRate": 9600,
    "arduino.useArduinoCli": true
```
`arduino-cli` version:
```
arduino-cli alpha Version: 0.18.1 Commit: b3cf8e19b801b2855ccd2a18c69c673315839231 Date:
```
Logs:
```
[Starting] Analyzing sketch '03-love-o-meter/03-love-o-meter.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
[Warning] Failed to generate IntelliSense configuration.
[Error] Analyzing sketch '03-love-o-meter/03-love-o-meter.ino': {"error":{"errno":"ENOENT","code":"ENOENT","syscall":"spawn /usr/local/bin/arduino-cli/Contents/MacOS/Arduino","path":"/usr/local/bin/arduino-cli/Contents/MacOS/Arduino","spawnargs":["compile","-b","arduino:avr:uno","--verbose","/Users/calebho/work/arduino/starter-kit/03-love-o-meter/03-love-o-meter.ino"]}}
```
Seems like the issue is `"syscall":"spawn /usr/local/bin/arduino-cli/Contents/MacOS/Arduino"`

colivier, 2021-04-20:
I also have the same issue

VSCode :
`Version : 1.55.2 (user setup)
Commit : 3c4e3df9e89829dce27b7b5c24508306b151f30d
Date : 2021-04-13T09:35:57.887Z
Electron : 11.3.0
Chrome : 87.0.4280.141
Node.js : 12.18.3
V8 : 8.7.220.31-electron.0
OS : Windows_NT x64 10.0.19041`

adiazulay, 2021-04-20:
If your c_cpp_properties.json file was created using v0.3.x you may have to delete it and let v0.4.x recreate it.
If you want to keep your c_cpp_properties.json from v0.3.x you can display IntelliSense auto-generation in the extension settings.

colivier, 2021-04-20:
Thanks @adiazulay for the reply

I am using version v0.4 of the plugin and have never used version v0.3.

For my part, on a new project the Arduino project's .vscode subdirectory does not contain the c_cpp_properties.json file, there is just the arduino.json file. On my previous projects the file was created and everything was ok. The issue is just for the new projects

mahyar1284, 2022-01-23:
I have the same problem

enwi, 2022-01-24:
I don't know if this is the case for you, but I had this issue whenever the `.ino` file was not named exacctly after the folder it was in.
So for example if your `.ino` file is called `test.ino` make sure it is inside a folder called `test`.

mahyar1284, 2022-01-24:
> I don't know if this is the case for you, but I had this issue whenever the `.ino` file was not named exacctly after the folder it was in. So for example if your `.ino` file is called `test.ino` make sure it is inside a folder called `test`.

it partially fixed the issue (❁´◡`❁). now most of the time it generates the intellisence configuration!!!

maxandriani, 2022-05-06:
Hello guys! I have the same issue w/ win10. I also set the debug level to "vervose" but I didn't get any aditional info about the error.

settings.json
```json
{
    "arduino.path": "C:\\arduino-cli\\bin",
    "arduino.useArduinoCli": true,
    "arduino.enableUSBDetection": true,
    "arduino.logLevel": "verbose"
}
```

arduino.json
```json
{
    "port": "COM3",
    "board": "arduino:avr:uno",
    "sketch": "arduino-blink-app.ino"
}
```
```
files:
C:\user\shhh\projects\arduino-blink-app\*
arduino-blink-app.ino
.vscode\*
```
```
Name: Arduino
Id: vsciot-vscode.vscode-arduino
Description: Arduino for Visual Studio Code
Version: 0.4.12
Publisher: Microsoft
VS Marketplace Link: https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino
---
Version: 1.67.0 (system setup)
Commit: 57fd6d0195bb9b9d1b49f6da5db789060795de47
Date: 2022-05-04T12:06:02.889Z
Electron: 17.4.1
Chromium: 98.0.4758.141
Node.js: 16.13.0
V8: 9.8.177.13-electron.0
OS: Windows_NT x64 10.0.19042
---
arduino-cli.exe Version: 0.13.0 Commit: 693a045
```
[image]

blaze-Youssef, 2022-07-28:
@maxandriani 
Same here I have just installed vs code and the extensions and I have this issue, and  I can't get more info about the problem, it just prints "[Warning] Failed to generate IntelliSense configuration.".
Any fix ?

blaze-Youssef, 2022-07-28:
I solved the issue by installing Arduino ide and using it instead of Arduino-cli and it worked perfectly.

Bludator, 2022-08-18:
Same here, oddly enough it passes every third try on 'blink example'

maxandriani, 2022-09-12:
> @maxandriani Same here I have just installed vs code and the extensions and I have this issue, and I can't get more info about the problem, it just prints "[Warning] Failed to generate IntelliSense configuration.". Any fix ?

I didn't. I did give up and install arduino IDE. 🤣

Pixel48, 2022-11-15:
Problem solved: Arduino sketches are folders, and main sketch file is defined by same name as containing folder - just change your main .ino file to match project folder or vice-versa and you're good to go

stas35r, 2023-01-21:
I had the same problem on Windows 11 x64 using arduino-cli and Arduino IDE v2. The following helped me.
1) Install arduino-cli (I use scoop, current version 0.29.0)
2) Add the arduino-cli path to the "arduino.path" property (I have this C:\\Users \\{user}\\scoop\\shims )
3) Set "arduino.useArduinoCli": true
4) Downgrade the extension version to 0.4.11